donovan6000/M33-Fio #89: Unhandled firmware error

Timelord73, 2016-02-21:
This is the error I am getting and I have tried the trouble shooting and have no other COMS open open to any and all suggestions.

Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: COM6
Changing monitoring state from 'Detecting serial port' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: AUTO SerialException: 'could not open port 'COM6': WindowsError(5, 'Access is denied.')' @ comm.py:_openSerial:1349 (hook m3dfio)

donovan6000, 2016-02-21:
This should be fixed in the newest M3D Fio version. I just released V0.27 about an hour ago.

Timelord73, 2016-02-21:
Do I have to download the whole program again or can I just get the updated addition.

donovan6000, 2016-02-21:
You should be able to update M3D Fio from inside of OctoPrint's plugin manager. Go to the 'Software Update' tab in OctoPrint's settings, and it should show that an update is available for M3D Fio.

Timelord73, 2016-02-21:
It is saying that is up to date at V 0.26 and no updates

Timelord73, 2016-02-21:
Got it to work. Is there a manual or user guide I can use to reference?

donovan6000, 2016-02-22:
There's no manual for M3D Fio. There's probably some manuals on using OctoPrint is you search around.

closed 2016-02-25 by donovan6000