donovan6000/M33-Fio #90: Calibrating before print can be disastrous if you don't monitor the calibration.

LogicalChaos, 2016-02-21:
With the new support for "calibration before print" enabled, the second print with that setting failed, cosmetically damaging the printer.  This is not a problem with M3D-Fio, rather the OctoPrint video shows the M3D calibration obviously not working, and when the print head touches down to being printing, it's apparent the calibration was 5mm (or more) too low.  After cleaning up and re-calibrating, it prints just fine again.  Robust is not what you often hear with this printer, but it came through that print just fine, if not a little scarred.  Emergency stop works well.

You may want to put a caution on that radio button, to be cautious walking away.  I've used the calibration many times manually, and have never seen this type of failure before.

[image]
[image]

donovan6000, 2016-02-21:
Alright, I added a caution next to the option and updated its tool tip to advise against leaving the printer unmonitored until after the print starts. Hopefully that doesn't happen too frequently. Their accelemoter is really sensitive, so it must have picked up some vibrations or something.
[image]

LogicalChaos, 2016-02-21:
Looks good.  Wouldn't of saved me anyway, as I already ignore the "Dangerous" warning of ignoring print dimensions limitation because "I know what I'm doing" (obviously).  Maybe someone will pay attention to it though.

closed 2016-02-21 by LogicalChaos