donovan6000/M33-Fio #91: Where is model editor located?

abbradar, 2016-02-22:
First -- I'm sorry for the very stupid question! I've been trying to find how to invoke Modify Model dialog for some time. From screenshot and description it looks like it can be opened when slicing a model, but I see only Modify Profile dialog (and also the first generic one). I've also looked around OctoPrint in general (M3D Fio settings, various tabs...) but haven't noticed even a mention. How can I open it? To be clear, I mean this window: https://raw.githubusercontent.com/donovan6000/M3D-Fio/master/images/model%20editor.png .

ra100, 2016-02-22:
Hi, model editor is opened after you modify slicing profile. First you choose slicing profile, than you can modify it and after that you can modify model.

abbradar, 2016-02-22:
Very strange, right after modifying a profile the model is just sliced and I don't have any following dialogs.

abbradar, 2016-02-22:
Just to clarify: after this dialog:
[image]
when I click Slice, all dialogs are closed and sliced model is added to the files list.

donovan6000, 2016-02-22:
Your web browsers doesn't support WebGL or doesn't have it enabled

Can you see a spinning cube [here](https://get.webgl.org/)?

abbradar, 2016-02-22:
Huh, indeed -- Firefox has disabled WebGL support for me for some reason. After resetting `*context*` variables in `about:config` as was suggested [here](https://bbs.archlinux.org/viewtopic.php?id=207637) the cube appeared.

I would suggest adding a small warning to the "Modify Profile" dialog, like "Warning: WebGL is not supported, Model Editor is not enabled" -- how about this?

donovan6000, 2016-02-22:
Sounds good to me. The next version will have a warning about skipping the model editor if WebGL is disabled/not detected
[image]

closed 2016-02-22 by donovan6000