donovan6000/M33-Fio #92: Saving M3D-Fio printer setting interrups printing

Nuffe71, 2016-02-22:
I had multiple windows open and forgot that I was printing. I modified plugin settings, specifically filament temperature setting. The following happened (terminal):

```
Send: N46874 G1 X59.748505 Y62.866495 Z3.982708 E2924.684026**42
Recv: ok 46874
Send: N46875 G1 X52.677437 Y69.937563 Z4.084241 E2924.902294**33
Recv: ok 46875
Send: N46876 M618 S41 T1 P100*85
Recv: ok Target address out of range
Send: N46877 M619 S41 T1*20
Recv: ok
Recv: T:190.2164
Recv: wait
Recv: T:186.7942
Recv: wait
Recv: T:183.4360
```

At least I think the saving caused this.

P.S. this was with 0.26. Does "Software Update" plugin support M3D-Fio automatically or do I need to have settings for that in config.yaml?

donovan6000, 2016-02-23:
Ok, the next version won't save the setting changes to the printer if the printer is currently printing.

Yeah, software update should work directly on M3D Fio. Just click the 'Check for updates now' button at the bottom of the 'Software Updates' tab in OctoPrint's settings and it should ask if you want to update M3D Fio.

donovan6000, 2016-02-25:
Fixed in V0.27.1

closed 2016-02-25 by donovan6000