donovan6000/M33-Fio #94: Advanced Settings section cannot be opened

abbradar, 2016-02-23:
[image]

Here, Advanced Settings is hidden for me and the arrow doesn't do anything.

donovan6000, 2016-02-23:
What web browser are you using? All of my development and testing happens in Google Chrome.

abbradar, 2016-02-23:
It's Firefox 44. Interface in other places works fine. Sadly I don't understand anything in web development so I don't even know where to start by myself -- if you have time, please take a look.

abbradar, 2016-02-23:
FWIW, Manual Settings works fine.

donovan6000, 2016-02-23:
Ok, it's fixed now in the [devel branch](https://github.com/donovan6000/M3D-Fio/tree/devel). You can install it using OctoPrint's plugin manager by installing a plugin from this URL: `https://github.com/donovan6000/M3D-Fio/archive/devel.zip`

For some reason the click event wasn't being recognized there but the mousedown event was, so it was an easy fix. Usually Firefox conforms to web standards much better than Chrome, so maybe I just wasn't seeing something.

abbradar, 2016-02-25:
It works, thanks.

closed 2016-02-25 by abbradar