donovan6000/M33-Fio #95: Resuming printing after a long pause

abbradar, 2016-02-24:
Hi,

First, my story. I was printing a model for some time when I realized that slicer did a bad job and printer attempted to make a "platform" starting extruding in place without a support right now, so all the ink just got down inside the model. I paused the job and began figuring out what can I do -- printer was running the fans and keeping the temperature meanwhile. Then, after a long time, it suddenly stopped and by the time I made some makeshift support it wasn't reacting to "Resume" button, nor to manual controls.

I've restarted the printer and found G-code which was running when the printer got paused in the log. I made an assumption that all coordinates in the code are absolute (they look like they are), so if I take a G-code script for a model and drop all commands before that one, it should continue right at that point. Unfortunately, the printer needed to find out its center first (not calibrate, but do the usual X-Y limits search), so it attempted to lower the extruder and damaged the model for good.

There are two things that might be improved which I've thought about:
a) Bug with the printer suddenly stopping after a long period of pause time.
b) Ability to disable X-Y center search before starting the print (maybe I've missed one?) and start right from place it is positioned now.
b*) If X-Y center search is needed after all, add an option to do it a bit higher than the extruder is currently positioned, that is: 1) go a bit up; 2) calibrate; 3) go down to the original place, first X-Y and then lower the head.

Thanks for reading this! I understand that this is an edge case, so consider this low-priority. Also, I'm very new to 3D-printing so there's a good chance I'm misunderstanding how G-code commands work.

donovan6000, 2016-02-24:
This could actually be related to the printer's firmware since it does automatically turn off the heater, motors, and fan when it's inactive for too long and will send the response, "System has been inactive for too long, heater and motors have been turned off". I should see if I can have it periodically send a command to the printer when it's paused to prevent this.

Every time the printer is powered on, it's only aware of its current Z position since it only saves that, and not its X, Y, or E position, in EEPROM. So in order for it to perform absolute or relative movements it needs to know its X and Y position, which is obtained by doing the X-Y homing that sets X to 54 and Y to 50. So the X-Y centering is not something that we shouldn't disable since the printer will be in an unknown state without it. It also helps make sure that a print starts off correctly since some things can mess up the printer's internal X and Y. A G92 E0 command is also run before a print starts to set the amount of extruded filament, E position, to 0.

What you were trying to do with continuing a partial print after restarting the printer isn't very easy to do. You'd have to get the printer's internal X, Y, Z, and E values to all be what they were at before to make sure that partial print continue like it should. So you'd have to disable M3D Fio's preparation pre-processor since that's what adds in the X-Y homing and G92 E0 stuff. Then run the following commands to move the extruder back into the correct position. You can obtain the last values by looking at the last command that was run before in the G-code file. Then you could start the print like you were doing with removing all the commands before the desired command.
```
G90; Put into absolute mode
G28; Home X-Y
G92 ElastEValue; Set E
G0 XlastXValue YlastYValue ZlastZValue F3000; Set X, Y, and Z
```

So by disabling M3D Fio's preparation pre-processor you have the ability to disable the X-Y homing. Moving up the extruder up a bit before performing the homing isn't going to change much since the extuder would still have gone into the partial print since M3D Fio's preparation pre-processor moves Z to 0.4 before running any commands from the G-code file.

abbradar, 2016-02-25:
A-ha, thank you very much for so detailed answer! I'll try this next time slicer fails ~_^. Leaving this open but reducing scope to just (a).

donovan6000, 2016-03-15:
Alright, this is fixed now. The printer can be paused forever as long as a client is connected to OctoPrint while it's paused. The firmware will turn off its heater, motors, and fan after being inactive for 15 minutes, but when a client is connected to OctoPrint it'll send a useless command, G4, to the printer every 3 minutes which causes the printer to remain active. Just left my printer paused to 40+ minutes and didn't have any problems.

closed 2016-03-15 by donovan6000