donovan6000/M33-Fio #96: Unable to connect to M3D printer

beth21762, 2016-02-25:
Octoprint is running, I'm logged in and I click connect to connect to printer and I get the following message.

Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: COM4
Changing monitoring state from 'Detecting serial port' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: AUTO SerialException: 'could not open port 'COM4': WindowsError(2, 'The system cannot find the file specified.')' @ comm.py:_openSerial:1349 (hook m3dfio)﻿

I have made sure the M3D software is not running, even checked in Task Manager and looked in M's for Micro or M3D and did not see anything.  I'm not sure if it would be under another name. 

I then have rebooted my laptop and same problem.

Please help me figure out what is using/connected to my printer and preventing M3D-fio from connecting.

abbradar, 2016-02-25:
This sounds akin to my own issue which was fixed in https://github.com/donovan6000/M3D-Fio/pull/84. Maybe catching `WindowsError` along with `OSError` will help...

beth21762, 2016-02-25:
I have tried several times and continue to get the same thing over and over.  Where do I capture these other errors at?

abbradar, 2016-02-25:
Try to install M3D-Fio from this url: https://github.com/abbradar/M3D-Fio/archive/devel.zip

I've included a patch which hopefully will help.

beth21762, 2016-02-25:
Do I need to un-install it first, or just install from there?

abbradar, 2016-02-25:
Unfortunately I don't use OctoPrint plugin management for installing so I don't know how it works. I think you can try both ways ~_^.

donovan6000, 2016-02-26:
@beth21762 You should uninstall it first before installing any of the devel versions. You can do that by going to the Plugin Manager tab in OctoPrint's settings and clicking the garbage can next to M3D Fio.

Then to install the devel version of M3D Fio, click the Get More button the bottom of OctoPrint's plugin manager and choose to install a plugin from a URL. If it asks you to to restart OctoPrint, then you can restart your computer to do so.

I did just make a change that fixed connecting to the printer on one of my Windows computer, so it's worth giving both of these versions of M3D Fio a try to see if either fixes your issue.
```
https://github.com/abbradar/M3D-Fio/archive/devel.zip
https://github.com/donovan6000/M3D-Fio/archive/devel.zip
```

beth21762, 2016-02-26:
I will try this when I get home from work.  I tried the other last night,  it connected once, but bed needed calibrated, I tried it never did anything, just sat spinning for over an hour.  So I rebooted and then it wouldn't connect again.  This time having problems at the baud rate.  

I always havev t o unplug the printer multiple times, therefore it needs to be calibrated.  Can I do this from the octoprint?

I will let you know how it goes tonight.

beth21762, 2016-02-26:
Also, what are the differences between the two that you listed?  Should I try the top one first or the bottom one?

donovan6000, 2016-02-26:
abbradar version adds exception handling for the WindowsError, and mine doesn't remove the serial timeouts until after the printer has been connected. Try the top one first then the bottom.

M3D Fio adds calibration button in OctoPrint's controls tab which allow you to calibrate the printer using OctoPrint.

abbradar, 2016-02-26:
Just to clarify -- I haven't done any actual testing for my branch because I don't have a Windows setup, so if it doesn't help please disregard the patch.

beth21762, 2016-02-27:
I have tried both links and still having trouble.  Now it is having problems with the Baud rate.

Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: COM3
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x3e885d0, open=True>(port='COM3', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 250000
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
Baudrate test retry: 3
Send: M110
Recv: ?B004
Baudrate test retry: 2
Send: M110
Recv: ?B004
Baudrate test retry: 1
Send: M110
Recv: ?B004
Baudrate test retry: 0
Send: M110
Recv: ?B004
Trying baudrate: 230400
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
Baudrate test retry: 3
Send: M110
Recv: ?B004
Baudrate test retry: 2
Send: M110
Recv: ?B004
Baudrate test retry: 1
Send: M110
Recv: ?B004
Baudrate test retry: 0
Send: M110
Recv: ?B004
Trying baudrate: 115200
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
Baudrate test retry: 3
Send: M110
Recv: ?B004
Baudrate test retry: 2
Send: M110
Recv: ?B004
Baudrate test retry: 1
Send: M110
Recv: ?B004
Baudrate test retry: 0
Send: M110
Recv: ?B004
Trying baudrate: 57600
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
Baudrate test retry: 3
Send: M110
Recv: ?B004
Baudrate test retry: 2
Send: M110
Recv: ?B004
Baudrate test retry: 1
Send: M110
Recv: ?B004
Baudrate test retry: 0
Send: M110
Recv: ?B004
Trying baudrate: 38400
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
Baudrate test retry: 3
Send: M110
Recv: ?B004
Baudrate test retry: 2
Send: M110
Recv: ?B004
Baudrate test retry: 1
Send: M110
Recv: ?B004
Baudrate test retry: 0
Send: M110
Recv: ?B004
Trying baudrate: 19200
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
Baudrate test retry: 3
Send: M110
Recv: ?B004
Baudrate test retry: 2
Send: M110
Recv: ?B004
Baudrate test retry: 1
Send: M110
Recv: ?B004
Baudrate test retry: 0
Send: M110
Recv: ?B004
Trying baudrate: 9600
Send: M110
Recv: ?B004
Baudrate test retry: 4
Send: M110
Recv: ?B004
Baudrate test retry: 3
Send: M110
Recv: ?B004
Baudrate test retry: 2
Send: M110
Recv: ?B004
Baudrate test retry: 1
Send: M110
Recv: ?B004
Baudrate test retry: 0
Send: M110
Recv: ?B004
Changing monitoring state from 'Detecting baudrate' to 'Closed'
Changing monitoring state from 'Closed' to 'Error: No more baudrates to test, and no suitable baudrate found.'
Connection closed, closing down monitor

donovan6000, 2016-02-27:
It looks like either M3D Fio wasn't installed or it's not being loaded by OctoPrint. Do you see an M3D Fio tab in OctoPrint's settings?

Also try running OctoPrint manually so we can see if it's generate any other errors. Kill the process pythonw.exe in the task manger to stop the currently running OctoPrint, and run the following command in the command prompt to run OctoPrint.
```
%SYSTEMDRIVE%\python\Scripts\octoprint.exe
```

abbradar, 2016-02-27:
Notice that this is still very similar to what I've reported in https://github.com/donovan6000/M3D-Fio/issues/83

With what version do you have bad rate problems -- mine, @donovan6000's, even vanilla one? Also the log would indeed be very helpful.

deferredreward, 2016-02-27:
Since it's so closely related I'll add my terminal and logs here. Fedora 23, firewalld is off. I've used three different versions of the plugin, the dev versions give the same result.

With M3D-Fio from OctoPrint plugin store (0.27.1):

Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: AUTO Exception: 'Couldn't reconnect to the printer' @ comm.py:_openSerial:1349 (hook m3dfio)
Changing monitoring state from 'Detecting serial port' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: AUTO Exception: 'Couldn't reconnect to the printer' @ comm.py:_openSerial:1349 (hook m3dfio)

With abbradar dev version (0.27.20160223):

Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: AUTO NameError: 'global name 'WindowsError' is not defined' @ comm.py:_openSerial:1349 (hook m3dfio)

Wtih donovon dev version :

Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: AUTO Exception: 'Couldn't reconnect to the printer' @ comm.py:_openSerial:1349 (hook m3dfio)

[octoprintm3d.txt](https://github.com/donovan6000/M3D-Fio/files/149427/octoprintm3d.txt)

beth21762, 2016-02-27:
Yes, the M3D fio shows in the settings.  

So I closed the python and ran the command that you told me to run and below are the messages received.  It is stuck on connecting.

From Terminal:
Connecting to: COM3
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x43d6370, open=True>(port='COM3', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110

From CMD Prompt:
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\Momma>%SYSTEMDRIVE%\python\Scripts\octoprint.exe
2016-02-27 13:55:59,542 - octoprint.server - INFO - Starting OctoPrint 1.2.9 (HE
AD -> master branch)
2016-02-27 13:55:59,732 - octoprint.plugin.core - INFO - Loading plugins from C:
\Users\Momma\AppData\Roaming\OctoPrint\plugins, C:\python\lib\site-packages\octo
print-1.2.9-py2.7.egg\octoprint\plugins and installed plugin packages...
2016-02-27 13:55:59,845 - octoprint.plugins.discovery - INFO - pybonjour is not
installed, Zeroconf Discovery won't be available
2016-02-27 13:55:59,845 - octoprint.plugin.core - INFO - Found 6 plugin(s) provi
ding 6 mixin implementations, 6 hook handlers
2016-02-27 13:55:59,859 - octoprint.filemanager.storage - INFO - Initializing th
e file metadata for C:\Users\Momma\AppData\Roaming\OctoPrint\uploads...
2016-02-27 13:55:59,875 - octoprint.filemanager.storage - INFO - ... file metada
ta for C:\Users\Momma\AppData\Roaming\OctoPrint\uploads initialized successfully
.
2016-02-27 13:55:59,875 - octoprint.plugins.softwareupdate - INFO - Loaded versi
on cache from disk
2016-02-27 13:56:00,532 - octoprint.util.pip - INFO - Found pip at C:\python\Scr
ipts\pip.exe, version is 7.1.2
2016-02-27 13:56:00,532 - octoprint.plugin.core - INFO - Initialized 6 plugin im
plementation(s)
2016-02-27 13:56:00,532 - octoprint.plugin.core - INFO - 6 plugin(s) registered
with the system:
|  CuraEngine (<= 15.04) (bundled) = C:\python\lib\site-packages\octoprint-1.2.9
-py2.7.egg\octoprint\plugins\cura
|  Discovery (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7.egg\o
ctoprint\plugins\discovery
|  M3D Fio (0.27.20160223) = C:\python\lib\site-packages\octoprint_m3dfio
|  Plugin Manager (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7.
egg\octoprint\plugins\pluginmanager
|  Software Update (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7
.egg\octoprint\plugins\softwareupdate
|  Virtual Printer (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7
.egg\octoprint\plugins\virtual_printer
2016-02-27 13:56:00,532 - octoprint.filemanager - INFO - Adding backlog items fr
om all storage types to analysis queue...
2016-02-27 13:56:00,548 - octoprint.filemanager - INFO - Added 1 items from stor
age type "local" to analysis queue
2016-02-27 13:56:00,548 - octoprint.server - INFO - Reset webasset folder C:\Use
rs\Momma\AppData\Roaming\OctoPrint\generated\webassets...
2016-02-27 13:56:00,562 - octoprint.server - INFO - Reset webasset folder C:\Use
rs\Momma\AppData\Roaming\OctoPrint\generated\.webassets-cache...
2016-02-27 13:56:00,752 - octoprint.plugins.discovery - INFO - Registered OctoPr
int instance on Momma-Laptop for SSDP
2016-02-27 13:56:00,752 - octoprint.plugins.pluginmanager - INFO - Loaded plugin
 repository data from disk, was still valid
2016-02-27 13:56:00,829 - octoprint.server - INFO - Listening on http://0.0.0.0:
5000
2016-02-27 13:56:00,875 - octoprint.server - INFO - Preemptively caching / for {
'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://localhost:5000/'}
2016-02-27 13:56:01,250 - octoprint.server.util.sockjs - INFO - New connection f
rom client: 127.0.0.1
2016-02-27 13:56:04,039 - octoprint.server.util.sockjs - INFO - Client connectio
n closed: 127.0.0.1
2016-02-27 13:56:05,934 - octoprint.server.util.sockjs - INFO - New connection f
rom client: 127.0.0.1
2016-02-27 13:56:25,687 - octoprint.printer.standard - ERROR - Exception while a
dding communication log entry
Traceback (most recent call last):
  File "C:\python\lib\site-packages\octoprint-1.2.9-py2.7.egg\octoprint\printer\
standard.py", line 141, in _sendAddLogCallbacks
    try: callback.on_printer_add_log(data)
  File "C:\python\lib\site-packages\octoprint_m3dfio\__init__.py", line 4270, in
 on_printer_add_log
    self._printer.get_transport().write("Q")
  File "c:\users\momma\appdata\local\temp\easy_install-ym50ws\pyserial-2.7-py2.7
-win32.egg.tmp\serial\serialwin32.py", line 295, in write
    raise writeTimeoutError
SerialTimeoutException: Write timeout

donovan6000, 2016-02-28:
@deferredreward It looks like you might not have read/write access to the printer. Do you have the [udev rule](https://github.com/donovan6000/M3D-Fio/blob/master/installers/Linux/90-m3d-local.rules) installed?

deferredreward, 2016-02-29:
Thanks! Some combination of that & the incidental reminder to actually add myself to the tty & baud? something groups as specified in the octoprint install instructions fixed it.

beth21762, 2016-02-29:
Any suggestions on my issue, still not working

donovan6000, 2016-02-29:
@beth21762 Does it ever manage to connect if you try connecting to it several times?

beth21762, 2016-02-29:
@donovan6000 it has in the past, but not recently.

donovan6000, 2016-03-02:
@beth21762 Try out M3D Fio V0.28 that I just released to see if it helps at all. I changed some stuff about the connecting procedure in the last few days.

beth21762, 2016-03-02:
I will try and let you know.  Will probably be tomorrow evening before I can though.

beth21762, 2016-03-03:
Do I use this same link https://github.com/donovan6000/M3D-Fio/archive/devel.zip or this link I found on the website https://github.com/donovan6000/M3D-Fio.git.  I'm thinking the 2nd link is the correct one, but I just want to make sure before I get home tomorrow and try it.

donovan6000, 2016-03-03:
The link is https://github.com/donovan6000/M3D-Fio/archive/master.zip

beth21762, 2016-03-03:
Same problem, still not connecting to the printer

beth21762, 2016-03-03:
Octoprint is running, but it won't connect like something else is using it, but I have rebooted and the m3d spooler is not running.

beth21762, 2016-03-12:
I did install the new version.  I am now trying my 1st print with it.  I have not rebooted and verified I could reconnect.  I will try that again tonight and update then.

beth21762, 2016-03-12:
Printer calibration is off.  I ran full bed calibration and it is stuck on saving back left off set. Been stuck on this for over an hour.  My first print had to be canceled because of calibration and now I can't get it calibrated properly to try printing again.

beth21762, 2016-03-13:
I think I got it calibrated.  I also rebooted and it took several tries and unplugging and plugging in th printer, but it eventually did connect to it again.

donovan6000, 2016-03-13:
@beth21762 At least you can connect after a few attempts with the new version. Does it show any errors when it fails to connect now? You can check by doing the same thing you did before with running OctoPrint manually. Or is it just displaying the dialog that says 'No Micro 3D printer detected. Try cycling the printer's power and try again?'
```
%SYSTEMDRIVE%\python\Scripts\octoprint.exe
```

The problem with it hanging on saving the back left offset is a known issue that I've been trying to solve, but it's difficult to repeat. There's a chance that it can happen any time it needs to get the printer's current Z position from the printer, which happens when saving any of the offsets or when changing the filament mid-print.

beth21762, 2016-03-13:
Yes, I was very happy that it did eventually reconnect.  The last thing I did that got it to reconnect was closed and ran the command you listed above to restart it and it worked.  The error is what having was both that I had before, no printer detected and then having trouble getting the baud rate.

beth21762, 2016-03-13:
It is telling me there is an update.  In the past the updates have always failed.  Should I do the update or not worry about it?

donovan6000, 2016-03-13:
That firmware update is to increase the max temperature from 285 degrees to 315 degrees. There's nothing else new in it, so you don't need to worry about it.

It's still giving the 'SerialTimeoutException: Write timeout' error in the terminal that you were getting before?

beth21762, 2016-03-13:
Yes it did give that message.  I closed python and restarted and then it connected after like 2 or 3 tries.

beth21762, 2016-03-13:
After 10 hours and 24 minutes of printing it just stopped printing.  No idea why

beth21762, 2016-03-13:
There is message in cmd that says warning could not send message to client 127.0.0.1: pop from an empty deque then says deque index out of range.

Also this print job when done in M3D software take around 7 hours.  Should it take longer here?

beth21762, 2016-03-13:
I'm so disappointed because it was printing perfectly.

donovan6000, 2016-03-16:
The version of M3D Fio I just released V0.30 does fix an issue that could potentially cause it to hang while printing. Not sure it that's what caused you problem, but it's worth updating to it.
```
https://github.com/donovan6000/M3D-Fio/archive/master.zip
```

Also can you post the entire cmd output when the printer doesn't let you connect?

beth21762, 2016-03-17:
I have installed. It also connected fairly easy this time.  So I will let you know if any problems come up over the next week or so.  Some times I have trouble getting the m3d software to recognize the printer, so it could be partly that too.  

I have noticed the time to complete a job is off, is it that way for everyone?

It did keep mentioning that there is m3d upgrade and asking me if I wanted to upgrade.  I continued to say no.  Is this something I should say yes and upgrade or continue to say no like I did?

donovan6000, 2016-03-18:
M3D Fio doesn't have any control over the displayed print times. The G-code being sent to the printer gets pre-processed before being sent to account for things like backlash and bed compensation. When on-the-fly pre-processing is enabled, this happens as the command is being sent so OctoPrint never knows about all the extra stuff being sent to the printer which causes its estimated print time to be pretty inaccurate.

You can disable on-the-fly pre-processing which will pre-process all uploaded and sliced G-code files when they're uploaded to OctoPrint and applies any changes directly to the G-code files. That will make the print time more accurate. However when on-the-fly pre-processing is disabled you'll need to reupload/reslice any files after changing any of the print settings for them to be applied.

The newest M3D Fio includes a modified version of M3D's latest official firmware V2016012501 which increases the max temperature from 285 degrees to 315 degrees. I've given this modified firmware the version number V2016012503 so that M3D Fio knowns that it has a newer firmware available and will ask the user if they want to update to it. You don't need to it update to it if you don't feel like it.

beth21762, 2016-03-18:
Thank you for the information!!  My next question is what is the best way to calibrate the bed from Octoprint/M3dFio?  I ask because my 1st print job was fine, it was a little hard to remove the support between the raft and the object but other than that it was perfect.  So I went to print again and the calibration was off and no filament was coming out on to the bed like before.  I could click extrude and filament came out, so there was no clog.  I never could get it right and needed to get to bed.  So I went back to the M3D software, recalibrated and it worked great from their software.  I'm still learning the Octoprint/M3DFio software so any help is greatly appreciated.

donovan6000, 2016-03-18:
M3D Fio has all the same calibration routines as M3D's official software. The 'Calibrate bed center Z0' button is equivalent to their generic calibrate bed, and the 'Calibrate bed orientation' button is equivalent to their calibrate drive system.

The 'Run complete bed calibration' button is designed to fully calibrate your bed. It can take about half an hour since it runs the 'Calibrate bed center Z0', 'Calibrate bed orientation', and calibrates each of the four corner offsets.

beth21762, 2016-03-19:
Should I adjust the back/right orientation or the offset?  I have calibrated and it isn't right.  I am having a really hard time getting the settings correct for it to print.

ra100, 2016-03-19:
@beth21762 best way is to run full calibration, which calibrates automatically bed center z0 and orientation values, than will guide you through corner offset values

punch69, 2016-03-19:
I had the same connection issues, I tried to reboot the M3D and the Rsb_pi without success, after disabling/enabling the M3D-Fio plug-in the M3D got connected !

beth21762, 2016-03-19:
I have ran the full calibration multiple times and it is not configured right.  When I test print the .4 border it is way off

beth21762, 2016-03-21:
Well my connecting problems seem to be gone, but again after 11 hours of printing it just stops.  So disappointed.

Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Users\Momma>%SYSTEMDRIVE%\python\Scripts\octoprint.exe
2016-03-20 09:35:13,243 - octoprint.server - INFO - Starting OctoPrint 1.2.9 (HE
AD -> master branch)
2016-03-20 09:35:13,509 - octoprint.plugin.core - INFO - Loading plugins from C:
\Users\Momma\AppData\Roaming\OctoPrint\plugins, C:\python\lib\site-packages\octo
print-1.2.9-py2.7.egg\octoprint\plugins and installed plugin packages...
127.0.0.1 - - [20/Mar/2016 09:35:13] "GET /sockjs/info HTTP/1.1" 404 -
2016-03-20 09:35:13,720 - octoprint.plugins.discovery - INFO - pybonjour is not
installed, Zeroconf Discovery won't be available
2016-03-20 09:35:13,721 - octoprint.plugin.core - INFO - Found 6 plugin(s) provi
ding 6 mixin implementations, 6 hook handlers
2016-03-20 09:35:13,756 - octoprint.filemanager.storage - INFO - Initializing th
e file metadata for C:\Users\Momma\AppData\Roaming\OctoPrint\uploads...
2016-03-20 09:35:13,828 - octoprint.filemanager.storage - INFO - ... file metada
ta for C:\Users\Momma\AppData\Roaming\OctoPrint\uploads initialized successfully
.
2016-03-20 09:35:13,838 - octoprint.plugins.softwareupdate - INFO - Loaded versi
on cache from disk
127.0.0.1 - - [20/Mar/2016 09:35:14] "GET /sockjs/info HTTP/1.1" 404 -
2016-03-20 09:35:14,780 - octoprint.util.pip - INFO - Found pip at C:\python\Scr
ipts\pip.exe, version is 7.1.2
2016-03-20 09:35:14,783 - octoprint.plugin.core - INFO - Initialized 6 plugin im
plementation(s)
2016-03-20 09:35:14,786 - octoprint.plugin.core - INFO - 6 plugin(s) registered
with the system:
|  CuraEngine (<= 15.04) (bundled) = C:\python\lib\site-packages\octoprint-1.2.9
-py2.7.egg\octoprint\plugins\cura
|  Discovery (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7.egg\o
ctoprint\plugins\discovery
|  M3D Fio (0.30) = C:\python\lib\site-packages\octoprint_m3dfio
|  Plugin Manager (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7.
egg\octoprint\plugins\pluginmanager
|  Software Update (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7
.egg\octoprint\plugins\softwareupdate
|  Virtual Printer (bundled) = C:\python\lib\site-packages\octoprint-1.2.9-py2.7
.egg\octoprint\plugins\virtual_printer
2016-03-20 09:35:14,789 - octoprint.filemanager - INFO - Adding backlog items fr
om all storage types to analysis queue...
2016-03-20 09:35:14,822 - octoprint.server - INFO - Reset webasset folder C:\Use
rs\Momma\AppData\Roaming\OctoPrint\generated\webassets...
2016-03-20 09:35:14,887 - octoprint.server - INFO - Reset webasset folder C:\Use
rs\Momma\AppData\Roaming\OctoPrint\generated\.webassets-cache...
2016-03-20 09:35:15,026 - octoprint.filemanager - INFO - Added 9 items from stor
age type "local" to analysis queue
2016-03-20 09:35:15,092 - octoprint.plugins.discovery - INFO - Registered OctoPr
int instance on Momma-Laptop for SSDP
2016-03-20 09:35:15,095 - octoprint.plugins.pluginmanager - INFO - Loaded plugin
 repository data from disk, was still valid
2016-03-20 09:35:15,203 - octoprint.server - INFO - Listening on http://0.0.0.0:
5000
2016-03-20 09:35:15,288 - octoprint.server - INFO - Preemptively caching / for {
'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://localhost:5000/'}
2016-03-20 09:35:17,234 - octoprint.server.util.sockjs - INFO - New connection f
rom client: 127.0.0.1
2016-03-20 09:35:22,848 - octoprint.server.util.sockjs - INFO - Client connectio
n closed: 127.0.0.1
2016-03-20 09:35:24,938 - octoprint.server.util.sockjs - INFO - New connection f
rom client: 127.0.0.1
2016-03-20 09:39:36,700 - octoprint.plugins.cura - INFO - Running u'C:/Program F
iles (x86)\\M3D - Beta Edition\\2016.01.26.1.4.0.9\\Resources\\CuraEngine\\CuraE
ngine.exe -v -p -s initialSpeedupLayers=4 -s minimalFeedrate=12 -s preSwitchExtr
uderCode=;Switch between the current extruder and the next extruder, when printi
ng with multiple extruders.\n;This code is added before the T(n)\n -s retraction
AmountExtruderSwitch=14500 -s supportXYDistance=700 -s insetXSpeed=14 -s retract
ionZHop=100 -s raftSurfaceLinewidth=350 -s postSwitchExtruderCode=;Switch betwee
[... 331 lines cut ...]
Send: N313631 G0 F1200 X60.449 Y55.251*113
Recv: ok 313631
Send: N313632 G0 X60.449000 Y55.251000 Z52.916694 F1200**14
Recv: ok 313632
Send: N313633 G1 F3000 X60.147 Y54.950 E8080.09268*12
Recv: ok 313633
Send: N313634 G1 F1200 E8077.69268*13
Recv: ok 313634
Send: N313635 G1 Z53.000*123
Recv: ok 313635
Send: N313636 G0 X60.050 Y50.398*30
Recv: ok 313636
Send: N313637 G1 Z52.900*113
Recv: ok 313637
Send: N313638 G1 F1200 E8080.09268*15
Recv: ok 313638
Send: N313639 G1 F3000 X59.573 Y50.168 E8080.11140*8
Recv: ok 313639
Send: N313640 G1 X59.907 Y50.750 E8080.13581*106
Recv: ok 313640
Send: N313641 G1 X60.395241 Y51.303458 Z52.874685 E8080.13581 F3000**113
Recv: ok 313641
Send: N313642 G0 F1200 X59.797 Y51.135*122
Recv: ok 313642
Send: N313643 G0 X59.797000 Y51.688458 Z52.879147 F1200**14

donovan6000, 2016-03-21:
The data being shown in OctoPrint's terminal tab isn't always what M3D Fio is sending/receiving so it doesn't help much to with fixing the issue where a print hangs. That's why in M3D Fio's settings you can now enable a logging sent/received data which logs everything.
[image]
If you can get the print to hang while that option is enabled then I can figure out what's causing it. Keep in mind that the log will clear itself every time a print starts, so make sure you download the log after immediately after the print hangs.

beth21762, 2016-03-22:
It won't let me attach file as it is too large.  I tried to copy and paste what is in the file and that isn't working either.  But I do have the log.

donovan6000, 2016-03-22:
If it's less than 300MB then you can upload it to [Send Space](https://www.sendspace.com/)

beth21762, 2016-03-22:
What email do I send it to?

donovan6000, 2016-03-22:
Don't use any emails. Just click upload and the next page will give you a link to the file that you can paste here.

beth21762, 2016-03-23:
https://www.sendspace.com/file/1a4zyp

donovan6000, 2016-03-23:
It's kinda weird that it never logs anything after resending a command, rs 46729, since the printer should always be periodically sending temperature readings and 'wait' responses which would get logged. This would indicate that either OctoPrint experienced an error that caused it to crash or that there's a bug in the printer's firmware that caused the printer to crash.

The cmd output you posted the other day about how the print stopped after 11 hours doesn't show any errors, so I'm really starting to feel like the problem is resulting from the printer's firmware. So I guess we should try seeing if using an older firmware will fix this problem. [This](https://github.com/donovan6000/M3D-Fio/files/185456/2016010812.zip) is the firmware that came with M3D's official software V1.4.0.7. Can you install that and do another printer see if the issue is resolved? Sorry for having you test so much stuff. I've been trying to get my printer to hang for a couple of days but have been unsuccessful.

beth21762, 2016-03-23:
No need for you to apologize.  I truly appreciate all of your help!!  The last couple of days I haven't been able to get any decent print jobs at all.  Calibration is all messed up.  I will try installing the old software tonight and test again to see what happens.

Also, it is normal for the print jobs through Octoprint/M3Dfio to take twice as long to print as it does through the m3d software?

beth21762, 2016-03-23:
That is hex file and my computer doesn't know what to do with it

beth21762, 2016-03-23:
I found the file to download from M3D and am installing it now.

donovan6000, 2016-03-23:
The hex file is meant to be used with M3D Fio's 'Update firmware from file' button. M3D's V1.4.0.7 software will install the same firmware, so it doesn't really matter which method you use.
[image]

The time of the print jobs is determined by how the model is sliced. OctoPrint sends commands to the printer at the same rate as M3D's official software, so that shouldn't be making a difference. You can try increasing the print speed when you slice to get faster results.
[image]

beth21762, 2016-03-24:
Any tips on calibrating?  I have ran all of the calibration methods in both softwares and can't seem to get anything that works properly.  Printing is even worse since I changed m3d software.  :-1:

donovan6000, 2016-03-24:
First, your 'Bed Height Offset' and 'External Bed Height' settings should both be 0mm, so zero them both out and click save in the windows that opens when you click the 'Print settings' button in the general controls section.

Then make sure you bed center Z0 is calibrated correctly. Run the following commands in OctoPrint's terminal tab to position the extruder to be at its lowest position in the middle of the bed. It should be barely touching the bed after that, but if it's not then you can fine tune the extuder's Z by using the controls in the movement controls section. After fine tuning the extruder's Z position, press the 'Save Z as bed center Z0' button to save the new Z0 value.
```
G28
G90
G0 Z0 F90
```
Be aware that running the G0 Z0 F90 command can cause the extruder to dig into the bed if your bed center Z0 is calibrated too low, so be cautious.

And finally make sure your calibrated offsets are correct. To do that press the diagonal arrow in the calibration controls section to move the extruder above one of the corners, then press the down arrow next to it which moves the extruder down to the currently saved offset for that corner. The extruder should be barely touching the bed if the offset for that corner is correct. If it's too high then adjust the extruder's Z using the buttons in the movement controls section, and press the 'Save Z as blah blah Z0' button to save the current Z as the new calibrated offset for that corner after your done fine tuning it. Then repeat that process for the other three corners.
[image]

Also M3D's official software V1.4.0.9 has been notorious for resetting the saved calibration offsets for no reason, so avoid using that version if your going to use their software for anything.

[11 more replies]

closed 2016-04-25 by donovan6000