donovan6000/M33-Fio #98: Shared library fails with an exception

abbradar, 2016-02-26:
For this G-Code generated with Slic3r: https://abbradar.net/me/share/untitled_M3D.gcode

Shared library terminates with an exception, crashing whole OctoPrint along with it.

abbradar, 2016-02-26:
I try now to run the same code with the shared library disabled; it takes veeeery long time (4th generation mobile Core i7). Let's see how it goes...

donovan6000, 2016-02-27:
It's fixed in the [devel branch](https://github.com/donovan6000/M3D-Fio/tree/devel) now. I've only ever seen the fan speed command, M106, use an S parameter to specify the speed, but your G-code is using a P parameter to do that.

abbradar, 2016-02-27:
From [this](http://reprap.org/wiki/G-code#M106:_Fan_On) it seems that P sets _what_ fan to use, not its speed. This command looks malformed because the speed is not specified; maybe just ignore them? I'll try to understand why Slic4r generates them.

donovan6000, 2016-02-27:
The G-code file's settings have the fan speed at 100%, so maybe whatever printer type Slic3r is set to use expects P instead of S.
```
; max_fan_speed = 100
; min_fan_speed = 100
```

M3D Fio attempts to detect the fan speed before printing so that it can generate a warning if printing without using the fan. It's doesn't affect anything else and it still lets the print proceed even with no fan speed, so it doesn't really matter much.

abbradar, 2016-03-09:
Let's close this, thanks for the fix!

closed 2016-03-09 by abbradar