donovan6000/M33-Fio #99: Loop on saving offset

ra100, 2016-02-28:
On saving offset I get command loop. Whan I copy value and use settings and save, everything seems fine.
```
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Recv: wait
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Recv: wait
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Recv: ok 0
Send: M114
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Recv: wait
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Recv: wait
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Recv: wait
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Send: M105
Recv: ok
Recv: T:0
Send: N0 M110
Send: M114
Recv: ok 0
Recv: ok  X:98.9937 Y:94.9949 E:0.0000 Z:1.4000
Recv: wait
Send: M105
Recv: ok
Recv: T:0
```

donovan6000, 2016-02-29:
Give the newest devel version a try to see if it fixed this.

ra100, 2016-02-29:
Just tried it, its fixed, thanks.

closed 2016-02-29 by ra100