donovan6000/M33-Fio #101: Feature: Save calibration values as "profiles"

ra100, 2016-02-29:
I've got 2 printbeds, with different surfaces (buildtak, glass), and they have different heights. (I'm experimenting a bit). So when I want to switch them, I need to do complete recalibration + some manual adjustments.
Maybe when I've always save offset and bed height values somethere, and then type them in manually, it would work. But after bed height calibration, I always get the height value 0, so I think this could be problem when copy-pasting +saving values.
It would be at least more convenient, to just save and load "offset profile". But I undestand, that it wouldn't make much sense to work on this feature just for one person, so I ask, is there someone else, who would use this?

donovan6000, 2016-02-29:
I don't have a problem with implementing something like this. It'll probably be useful too for people with multiple printers.

donovan6000, 2016-03-01:
Alright, this is implemented now. There's two buttons in the calibration section called 'Save printer settings to file' and 'Restore printer settings from file' that allow you to save/restore the printer settings easily.

ra100, 2016-03-01:
awesome, thanks

closed 2016-03-09 by ra100