donovan6000/M33-Fio #107: At print end, head didn't stop moving and extruding

ra100, 2016-03-05:
I've let printer on during night and this is what I found, I think, that print was done for abou 7 hours. I hope my printer is no damaged. I've heard how head tried to move to the corner so I unplugged it
[image]
[image]
[serial.4.log.zip](https://github.com/donovan6000/M3D-Fio/files/159736/serial.4.log.zip)

log around time prin ended, I think.
```
2016-03-05 01:37:16,946 - SERIAL - DEBUG - Send: N64554 G1 X61.230000 Y52.945900 Z51.864477 E3424.03983 F840**122
2016-03-05 01:37:16,957 - SERIAL - DEBUG - Recv: ok 64554
2016-03-05 01:37:16,966 - SERIAL - DEBUG - Send: N64555 G1 X61.114 Y52.221 E3424.05885*81
2016-03-05 01:37:16,987 - SERIAL - DEBUG - Recv: ok 64555
2016-03-05 01:37:17,017 - SERIAL - DEBUG - Send: N64556 G1 X60.933 Y52.536 E3424.07736*90
2016-03-05 01:37:17,038 - SERIAL - DEBUG - Recv: ok 64556
2016-03-05 01:37:17,044 - SERIAL - DEBUG - Send: N64557 G1 X60.275 Y53.262 E3424.12726*80
2016-03-05 01:37:17,065 - SERIAL - DEBUG - Recv: ok 64557
2016-03-05 01:37:17,070 - SERIAL - DEBUG - Send: N64558 G1 X60.275 Y62.738 E3424.60987*87
2016-03-05 01:37:17,137 - SERIAL - DEBUG - Recv: ok 64558
2016-03-05 01:37:17,141 - SERIAL - DEBUG - Send: N64559 G1 X60.275000 Y58.341900 Z51.954809 E3424.330979**24
2016-03-05 01:37:17,150 - SERIAL - DEBUG - Recv: ok 64559
2016-03-05 01:37:17,154 - SERIAL - DEBUG - Send: N64560 G1 X60.275000 Y60.341900 Z52.008142 E3424.432838**29
2016-03-05 01:37:17,172 - SERIAL - DEBUG - Recv: ok 64560
2016-03-05 01:37:17,181 - SERIAL - DEBUG - Send: N64561 G1 X60.275000 Y63.817900 Z52.100836 E3424.60987**34
2016-03-05 01:37:17,193 - SERIAL - DEBUG - Recv: ok 64561
2016-03-05 01:37:17,197 - SERIAL - DEBUG - Send: N64562 G1 X60.933 Y63.464 E3424.65977*86
2016-03-05 01:37:17,231 - SERIAL - DEBUG - Recv: ok 64562
2016-03-05 01:37:17,234 - SERIAL - DEBUG - Send: N64563 G1 X61.233000 Y64.543900 Z52.121365 E3424.65977 F840**118
2016-03-05 01:37:17,248 - SERIAL - DEBUG - Recv: ok 64563
2016-03-05 01:37:17,261 - SERIAL - DEBUG - Send: N64564 G1 X61.114 Y63.779 E3424.67827*85
2016-03-05 01:37:17,277 - SERIAL - DEBUG - Recv: ok 64564
2016-03-05 01:37:17,284 - SERIAL - DEBUG - Send: N64565 M105*35
2016-03-05 01:37:17,717 - SERIAL - DEBUG - Recv: T:247.8935
2016-03-05 01:37:18,089 - SERIAL - DEBUG - Recv: ok 64565
2016-03-05 01:37:18,098 - SERIAL - DEBUG - Send: N64566 G1 X61.230 Y64.134 E3424.69729*85
2016-03-05 01:37:18,139 - SERIAL - DEBUG - Recv: ok 64566
2016-03-05 01:37:18,148 - SERIAL - DEBUG - Send: N64567 G1 X61.268 Y64.499 E3424.71598*90
2016-03-05 01:37:18,173 - SERIAL - DEBUG - Recv: ok 64567
2016-03-05 01:37:18,183 - SERIAL - DEBUG - Send: N64568 G1 X61.230 Y64.865 E3424.73472*80
2016-03-05 01:37:18,249 - SERIAL - DEBUG - Recv: ok 64568
2016-03-05 01:37:18,254 - SERIAL - DEBUG - Send: N64569 G1 X61.230000 Y65.944900 Z52.159253 E3424.73472 F840**113
2016-03-05 01:37:18,268 - SERIAL - DEBUG - Recv: ok 64569
2016-03-05 01:37:18,273 - SERIAL - DEBUG - Send: N64570 G1 X61.114 Y65.220 E3424.75374*81
2016-03-05 01:37:18,304 - SERIAL - DEBUG - Recv: ok 64570
2016-03-05 01:37:18,335 - SERIAL - DEBUG - Send: N64571 M105*38
2016-03-05 01:37:20,776 - SERIAL - DEBUG - Recv: T:242.1071
2016-03-05 01:37:21,770 - SERIAL - DEBUG - Recv: T:242.1071
2016-03-05 01:37:23,768 - SERIAL - DEBUG - Recv: T:242.1071
```

donovan6000, 2016-03-05:
Wow, that turned out bad. Hopefully your printer's not damaged. Unfortunately I can't tell much from the log of what went wrong.

The validation pre-processor will remove M105 commands since the printer constantly send its temperature anyway, and it'll send G4 command in place of sending a removed command since it has to send something to the printer to make sure all the line numbers being sent by OctoPrint stay correct. It did handle all the M105 commands correctly up until the point where it stopped, N64571, so that's probably not what caused this.

N64571 isn't one of the line numbers that affects other things, like N0 and N65535, so the line number probably didn't cause this.

Behind the scenes, M3D Fio handles resending commands that are requested from the printer to be resent. I couldn't let OctoPrint handle that anymore since the newer firmwares, starting with V2015122112, are capable of processing commands out of order which gets messy when the line numbers roll over from N65535 to N0. M3D Fio also limits the amount of commands that can be sent to the printer to one at a time, and it will also remember the last line number sent. So if it ever receives a 'wait' response, it'll convert that to a resend the last command response to prevent issues where M3D Fio would hang forever if a command was ignored by the printer.

In your case, it looks like something was preventing the printer from confirming that the N64571 command was ran, ok N64571. The printer will automatically turn off its heater and fan after being inactive for some period of time, like 10 minutes, but your printer didn't do this so we can conclude that M3D Fio was still sending stuff to the printer or a bug in the printer's firmware caused it to hang. What's really weird about this is that there's nothing being sent by the printer, except temperature readings, after it receives the N64571 command. If it was sending a 'wait' response, M3D Fio would resend the N64571 command and that response would be logged as an empty line 'Recv: ' in the serial log, but that never happened. So I think this is the result of a bug in the firmware.

ra100, 2016-03-05:
Well, printer seems to be OK, I recalibratet center Z0 and tried test border, no problem there.
And as it seemed like fw problem. I'll close it. Thanks.
And well, I just can say, that result of this error is some kind of art :D

closed 2016-03-05 by ra100