donovan6000/M33-Fio #108: Feature: Set number of brim lines

ra100, 2016-03-05:
I know there was maybe in profiles V3, that I could modify number of brim lines and I haven't seenit there or some time. I'd like to use brim, but with large model I get message, that model won't fit, I think it's because model + brim has first layer too large for printbed.
And another thing, would it be possible to visualize brim/raft in model editor? I understand, that brim/raft are added when gcode is being generated. Maybe some extra bounding box would be enough.

donovan6000, 2016-03-05:
All the settings shown in the profile editors advanced section are not all the possible settings. The VX profiles were profiles exported from Cura, so they had all the possible settings. The new profiles are converted directly from the Cura Engine parameters generated by M3D's official software, so they only include settings that were used by those parameters. You can always add in the line `brim_line_count = 20` to the advanced section to set the brim line count manually. I'll add it to the profiles though since it makes sense to have it now that there's an option to use a brim.

Visualizing the brim/raft will be difficult, and I'm not really sure how much more time I want to put into M3D Fio.

ra100, 2016-03-05:
Thanks. And as I thought, visualisation would take much more time than the gain would be, probably. Maybe I'll look into it, when I'll learn how stl viewer works, and what can be done with it.

donovan6000, 2016-03-05:
Alright, the brim line count setting is implemented in the devel branch now. It'll display the raft airgap setting if raft is enabled, and it'll display the brim line count is brim is enabled.

The part about visualizing the raft/brim that has me stumped is how to scale it. For instance, I duplicated this dragon and set the height of that clone to 0 to make it look like a raft. So we'd just have to scale it's X and Z up to make it wider than the model to make it look like a real raft/brim.
[image]
However scaling is done from the objects center, so it wont work correctly for complex objects like this. You can see this by looking at the dragon's back foot which now has no raft underneath it after scaling the raft.
[image]

ra100, 2016-03-05:
Wow, I am amazed :) I think, that scaling that doesn't exacatly match brim isn't much of a problem, it will still show you approx. boundaries for longer dimension of model and even in this format it would help with better fitting on printbed.

donovan6000, 2016-03-08:
The newest devel version has visual rafts/brims now. The raft should be accurate since its length on each side is defined by the Cura parameter raft_margin, however I'm not sure how accurate the brim is since I'm assuming its length on each side is brim_line_count * bottom_thickness.

The bottom boundary tier also accounts for the raft/brim size, so they'll turn red if the raft/brim goes out of bounds. All the other operations like cloning/cutting/merging/selecting should also work fine still. The only problem I'm still having is getting the raft/brim shape to change when model is rotated on its X or Z axis since I have to scale to raft/brim's Y to 0 on the world's Y axis which isn't easy to do with threejs.

donovan6000, 2016-03-08:
Managed to fix the problem I was having. It's all working now :)

[image]

ra100, 2016-03-09:
Nice job, you've done it sooner than I managed to suggest something :)

closed 2016-03-15 by donovan6000