donovan6000/M33-Fio #109: Gcode layer temperature / OctoPrint Target temperature/ Offset temperature not working.

soimy, 2016-03-07:
I'm using Simplify3D for STL slicing. Profile sets to Layer 0: 230C, Layer 1: 190C.
I can see on octoprint Temperature panel, the target temperature curve changing, but does no effect on M3D's actual temperature.
Seting target temperature / offsest temperature has no effect too.

The only way to tweak nozzle temperature is through Control tab -> Extruder Temperature.

donovan6000, 2016-03-07:
If you have M3D Fio's validation pre-processor's remove temperature commands option enabled, then all temperature commands in the G-code file wont be sent to the printer. If your using M3D Fio's on the fly pre-processing, then OctoPrint will still think that the temperature commands are being sent to the printer which is why the target temperature will be set to that value. If you have M3D Fio thermal bonding pre-processor enabled, then the first layer will be set to slightly higher than M3D's temperature setting and the second layer to be exactly M3D's temperature setting. The tooltips on all the settings explain this.

So you should disable the validation pre-processor's remove temperature commands option and the thermal bonding pre-processor if your trying to print a file that contains temperature command that you want to use.

The only ways to set the temperature during a print when using on the fly pre-processing and the validation pre-processor's remove temperature commands option are from the Control tab or from the Terminal tab. I'll look into modifying the Temperature tab's set target and set offset so that it can also work under those circumstances.

donovan6000, 2016-03-07:
The newest devel version will now let you set the temperature from the Temperature tab. You can install it using OctoPrint's builtin plugin manager by installing a plugin from `https://github.com/donovan6000/M3D-Fio/archive/devel.zip`.

soimy, 2016-03-08:
Great works @donovan6000 !

closed 2016-03-15 by donovan6000