donovan6000/M33-Fio #110: Running behind a proxy

abbradar, 2016-03-07:
I run OctoPrint behind an nginx proxy, so I have external URL `http://abc.xyz/printer` which is redirected to OctoPrint. It works fine with OctoPrint itself -- all assets are loaded with the right absolute path (e.g. `http://abc.xyz/printer/static/img/apple-touch-icon-114x114.png`). On the other hand, M3D-Fio doesn't respect the absolute path, so it tries to load assets from e.g. `http://abc.xyz/plugin/m3dfio/static/img/graph%20background.png` (notice absence of `/printer`). The absolute path is forwarded to OctoPrint via `X-Script-Name` header, which is `/printer` in this case -- its use is documented in OctoPrint manual.

donovan6000, 2016-03-07:
Try using the newest devel version, `https://github.com/donovan6000/M3D-Fio/archive/devel.zip`, and let me know if this is fixed. I switched all the absolute URLs to use the PLUGIN_BASEURL variable defined by OctoPrint which should account for X-Script-Name header.

abbradar, 2016-03-09:
It seems not all things were fixed, I've caught another one: `http://abc.xyz/plugin/m3dfio/download/profile_632412m3d_pla`

donovan6000, 2016-03-09:
Alright, all the URLs should be working now.

abbradar, 2016-03-09:
Seems that it works now. Highly appreciated, as always!

closed 2016-03-09 by abbradar