donovan6000/M33-Fio #111: User management fails

ThijsRoumen, 2016-03-07:
I installed Octoprint and set up a user account in the browser app. However when I install the M3DFio plugin the whole user management seems to be gone/changed. Every time you try to click login or write in one of the textfields the login UI disappears. I tried many ways but did not manage to login normally. 

As a consequence with access control enabled there is no chance to install the M3D plugin. How did you guys do this? 

I am running Octoprint (1.2.9) and M3DFio  from a Ubuntu system (UDOO)

donovan6000, 2016-03-07:
Sounds like there's an issue with M3D Fio's javaScript. Can you right click anywhere in the web browser and click on the inspect option. That'll open another that'll show any JavaScript errors that the web browser detected. What does it show?

ThijsRoumen, 2016-03-07:
Thanks for your quick response! It seems that JS is indeed the issue, at least that would be most likely from the symptoms observed. When I run Octoprint itself there are no errors. With the plugin I get one error message: 
Uncaught Syntax Error: unexpected identifier in packed_app.js:6605

donovan6000, 2016-03-07:
Can you click on the packed_app.js:6605 text next to it which will open the JavaScript code for the page. It it doesn't automatically go to line 6605, then manually scroll there. What's on that line.

ThijsRoumen, 2016-03-07:
nothing which seems relevant:
uniform vec3 viewVector;

within the definition of vertexShader:
var vertexShader = ' 
 uniform vec3 viewVector
 uniform float c;
 uniform float p
 ....

donovan6000, 2016-03-07:
It might be because I'm using grave accents ` to enclose sections of text. This is a new feature in ECMAScript V6, so make sure your using the newest version of Google Chrome since your vesion might not support it. I'm using V48.0.2564.82 and it's working well.

ThijsRoumen, 2016-03-07:
thanks for the suggestion. I will try it tomorrow, unfortunately I just got kicked out of the building because security closed it down :(

ThijsRoumen, 2016-03-08:
Thanks a lot, this was indeed the problem. I now connected to my Udoobuntu system via a server and ran it from my laptop and all was good. Luckily my goal was anyways to control the printer externally so you made my day :)

ThijsRoumen, 2016-03-08:
P.S. my version of Chrome was 18 so it was reasonably behind :P that was the one pre-installed on Udoobuntu1.1 for future reference

closed 2016-03-08 by ThijsRoumen