donovan6000/M33-Fio #112: Change max. temperature limit

ra100, 2016-03-09:
Is there way how could I set more than 285°C for heater? I tried it via Terminal, but printer's fw has probably some safety limits, or is it m3d fio thing? I got message that temperature has to be higher than 100 and less than 285 degrees.
Thing is, that I've got probably some hw problem, and I have to print at about 20-30 degrees higher than I should. But it would mean I shoud print with ABS at about 300 degrees. I've tried 285 and layers didn't stick very good together and it didn't look good at all. There is my motivation in trying to print at higher temp.

donovan6000, 2016-03-09:
That limitation is implemented in the printer's firmware and in the software, so increasing the temperature to more than 285 degrees would require changing both.

Give the newest devel version a try. It increases the max temperature to 350 degrees. It'll ask you to install the V2016012502 firmware which you'll need to use to get past the 285 degree limit.

ra100, 2016-03-09:
Thanks, it works, and at 305 abs extrudes quite well. Thanks this I'll be able to print with ABS after all :)

closed 2016-03-09 by ra100