donovan6000/M33-Fio #114: Add option to skip 3D model preview

abbradar, 2016-03-12:
Usecase: a very big and detailed model, which eats ~3GB RAM in the viewer (and a minute or so to load), and that's on a pretty high-end machine! That's a model fault (it's really detailed, for example Blender also takes time to load it).

donovan6000, 2016-03-13:
Ok, the devel branch now allows skipping the model editor.
[image]

beth21762, 2016-03-13:
Can I just say you are awesome!!!  Your quick responses to fixing issues and request for everyone is just unbelievable!!  I truly appreciate all of the help you have given me so far and that I know I will be asking for more in the future.  ;)

abbradar, 2016-03-13:
I very much second @beth21762. ~_^ Thanks for quick responses and for your commitment to fix issues.

donovan6000, 2016-03-15:
Thanks everyone! Glad I can help :)

closed 2016-03-15 by donovan6000