donovan6000/M33-Fio #119: Cannot connect to other printer types with M3D-Fio installed

LogicalChaos, 2016-03-23:
With the help of my M3D, I now have another printer set up.  But, I can only connect to it when I remove the M3D-Fio plugin.  I've created a second instance via the plugin, doesn't help.  Thoughts?

`Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x3118ad0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1532
Changing monitoring state from 'Connecting' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1532'
Connection closed, closing down monitor
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x31b1230, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1532
Changing monitoring state from 'Connecting' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1532'
Connection closed, closing down monitor`

donovan6000, 2016-03-23:
It should be able to connect if you enable the 'Not using a Micro 3D printer setting'. Each OctoPrint instance has its own independent settings, so changing that setting wont affect any other OctoPrint instances.
[image]
 I don't have another printer to actually test it with though so let me know if you still run into issues.

LogicalChaos, 2016-03-23:
Ah, thanks.  Didn't think of looking in on the plugin settings. Power outage now that the blizzard is over (sigh), so I'll check sometime later

LogicalChaos, 2016-03-24:
Well, I can connect now, but not actually send anything manually or through the controls.  Nothing happens.  No octoprint.log info, no activity on the printer.

`Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x1f52790, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: start
Recv: echo:Marlin1.0.2
Send: N0 M110 N0*125
Connection closed, closing down monitor
Changing monitoring state from 'Connecting' to 'Closed'
Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x211a530, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: FRMWARE_NAME:Marlin V1.0.2; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Wilson II EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: M21
Recv: echo:SD card ok
Recv: ok
Send: M20
Recv: Begin file list
Recv: CALIB~1.GCO
Recv: ROCKET~1.GCO
Recv: echo:Cannot open subdir
Recv: 1/
Recv: CONE~1.GCO
Recv: PLA_T~1.GCO
Recv: End file list
Recv: ok
Send: M105
Recv: T:17. /0.0 B:18.9 /0.0:1.9 /0.0 @:0 B@:0`

donovan6000, 2016-03-24:
M3D Fio appends asterisks, *, to the end of most commands sent by the buttons and all commands sent manually since that's used to tell the on-the-fly pre-processor not to process that command. That might be causing an issue here. Does OctoPrint's terminal tab show anything when you press the 'Absolute mode' or 'Relative mode' buttons? Those buttons don't get asterisks appended so they should get sent to the printer. And does the terminal tab show anything when you press the 'Home Z' button?

Also can you start a print?

LogicalChaos, 2016-03-26:
I've attached three logs from the serial.log.
[with_fio_disabled.txt](https://github.com/donovan6000/M3D-Fio/files/190556/with_fio_disabled.txt) is the normal startup sequence with M3D-Fio disabled.
[with_fio_enabled_connected.txt](https://github.com/donovan6000/M3D-Fio/files/190554/with_fio_enabled_connected.txt) is an attempt that connected initially, which succeeded, and then trying to send a G90 via the 'Absolute mode' button.  30 seconds after sending the command, it 'receives' a single character.  Looks suspiciously like a timeout.
[with_fio_enabled_typical.txt](https://github.com/donovan6000/M3D-Fio/files/190558/with_fio_enabled_typical.txt) is two connection attempts that are typical, which fail 90% of the time.  The first set I pressed the disconnect button, the second set I let it go for a while, as in I'm typing this, and it's still not generated another log line, 10 minutes later.

donovan6000, 2016-03-27:
Alright, try the latest devel version of M3D Fio. It'll now use OctoPrint's native connection routine when connecting to a printer when the 'Not using a Micro 3D printer' setting is enabled.

LogicalChaos, 2016-04-01:
Sorry for the delay.  Still no joy.  Pulled the devel version right now.
```
Connecting to: /dev/ttyUSB1
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6eda05b0, open=True>(port='/dev/ttyUSB1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Changing monitoring state from 'Connecting' to 'Closed'
Connecting to: /dev/ttyUSB1
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6e355670, open=True>(port='/dev/ttyUSB1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Connection closed, closing down monitor
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1532
Changing monitoring state from 'Connecting' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1532'
Connection closed, closing down monitor
```

donovan6000, 2016-04-03:
I'm not really sure why it's not working. There's nothing more I can actually do to connecting procedure since it's already [switching to use OctoPrint's native connecting procedure](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L8541-L8544) when the 'Not using a Micro 3D printer' option is enabled. Try the newest devel version, 0.31.2016040219, to see if that helped at all.

donovan6000, 2016-05-27:
It's been awhile since this issues had a response, so I'll probably close it in a couple of days. Let me know if you still have any interest in resolving this issue.

LogicalChaos, 2016-05-30:
At this time, I'm not going to pursue this.  Thanks.

closed 2016-05-30 by LogicalChaos