donovan6000/M33-Fio #120: Model editor not showing

ra100, 2016-03-24:
Before print after setting print options model editor doesn't show (in Opera).
In chrome plugins doesn't run at all, javascript crashes at ILLEGAL CHARACTER after var glowVertexShader =

donovan6000, 2016-03-24:
This is related to #111. ECMAScript V6 introduced using grave accents to enclose template literal strings in 2015, and I used it pretty frequently in M3D Fio since it makes code more readable, more manageable, and easier to reuse.

Here's what we used to have to do before  grave accents were introduced. It doesn't look too bad, but we also didn't have to escape any characters.
```
var glowVertexShader =
	"uniform vec3 viewVector;\n" +
	"uniform float c;\n" +
	"uniform float p;\n" +
	"varying float intensity;\n" +
	"void main() {\n" +
		"vec3 vNormal = normalize(normalMatrix * normal);\n" +
		"vec3 vNormel = normalize(normalMatrix * viewVector);\n" +
		"intensity = pow(c - dot(vNormal, vNormel), p);\n" +
		"gl_Position = projectionMatrix * modelViewMatrix * vec4(position, 1.0);\n" +
	"}\n"
;
```
And the same code when using grave accents.
```
var glowVertexShader = `
	uniform vec3 viewVector;
	uniform float c;
	uniform float p;
	varying float intensity;
	void main() {
		vec3 vNormal = normalize(normalMatrix * normal);
		vec3 vNormel = normalize(normalMatrix * viewVector);
		intensity = pow(c - dot(vNormal, vNormel), p);
		gl_Position = projectionMatrix * modelViewMatrix * vec4(position, 1.0);
	}
`;
```
Grave accents are still kinda new, so I guess Opera doesn't support it yet. Even though it'll fix compatibility with older browsers, I don't really want to switch to the old style.

ra100, 2016-03-25:
Sorry, my bad, I needed to restart Opera, maybe some problem with webgl, and I've had very very old chrome, so it didn't support ES6.

closed 2016-03-25 by ra100