donovan6000/M33-Fio #121: Print "fails" after first layer

ra100, 2016-03-24:
After first layer is done, head starts to move very slowly to FR corner, still extruding.
stl+gcode: [pin.zip](https://github.com/donovan6000/M3D-Fio/files/188768/pin.zip)

donovan6000, 2016-03-25:
I'm printing it right now. At layer 5 so far and I'm not experiencing any problems. Are you able to repeat this issue?

I'm really starting to think there's something wrong with M3D's V2016012501 firmware. Even M3D's tech support is now recommending that people downgrade to their V1.4.0.7 software which includes their V2016010812 firmware. I'll probably include that firmware in the next version of M3D Fio unless M3D release a new update in that time.

ra100, 2016-03-25:
Interesting, I tried it with 2 different models, this is result of first print, I didn't watch it and canceled print after 10 minutes or so. I'll just try flash firmware and run full calibrantion, as I see there is also external bed height calibration now. So at least I'll try that and report back.
[image]

ra100, 2016-03-25:
Seems everything is fine again. It was maybe just some glitch in my system.

ra100, 2016-03-25:
Seems like problem is with Wave Bonding, now I tried again with wave bonding, and again, after first layer of raft, head started to move to FR. And it also didn't seem like there were any "waves".
[log020160325.zip](https://github.com/donovan6000/M3D-Fio/files/189555/log020160325.zip)

donovan6000, 2016-03-25:
What other settings do you have enabled/disabled? I'm trying to recreate it, but I only got a successful print. The log does show that it moved into the FR corner from commands inserted by the pre-processors, so this is definitely the pre-processors fault.

Do you also expirence this issue in M3D Fio's V0.30.2 release? The only thing I've changed about the pre-processors since releasing that version was making sure that all commands have feedrates. I don't really think that would cause this though since the printer is already internally be using the last commands's feedrate if the current command doesn't have one.

ra100, 2016-03-26:
Now I noteiced, that Volume setting is somewhat weird. Shouldn't it be 109x113x116?
And I didn't try wave bonding for some time (it's probably not even worth it while printing on glass) so I'll have to try older version and try printing again.

[image]
[image]
[image]
[image]
[image]

ra100, 2016-03-26:
Same result with 0.30.2

donovan6000, 2016-03-26:
M3D Fio dynamically changes the dimensions of the M3D printer profile, micro_3d, whenever the boundaries of the printable region change. That happens when changing the external bed height or changing the external bed expands printable region setting, so that makes the dimensions in the printer profile always equal to the [size of the lowest boundary](https://github.com/donovan6000/M3D-Fio/blob/devel/octoprint_m3dfio/__init__.py#L758-L760). The only thing I've noticed that actually uses the printer profile dimensions is OctoPrint's G-code viewer to size the grid, so it's purely aesthetic.

I'm able to get the same unexplained slow FR movements when using wave bonding with all your settings. Turns out that there was an issue with the wave bonding and thermal bonding pre-processors where incrementing a layer counter could get messed up by the 'Use GPIO' and 'Mid-print filament change layers' settings. It should be fixed now.

ra100, 2016-03-29:
I've had time to test it just now. And it works nice, and it makes "waves". Thanks for fixing it.

closed 2016-03-29 by ra100