donovan6000/M33-Fio #123: Manual and advanced slicing settings are unusable

LogicalChaos, 2016-03-26:
Running Safari Version 9.0.3 (11601.4.4).  See screenshot.  The space for the two mentioned areas are virtually unusable.  The manual settings can be tabbed through to change then, the advanced settings text box can be 'seen', but barely.  The popup window cannot be resized to expose them further.
[image]

donovan6000, 2016-03-27:
Right now I'm only maintaining M3D Fio's compatibility with Google Chrome and Firefox. I don't have any computers that can run OS X >= 10.9 which is required for using the newest version of Safari, so there's not much I can do myself to make everything in M3D Fio compatible with it.

LogicalChaos, 2016-03-27:
Ahhh, launching in Chrome, I noticed to the left of the text are dialog expansion buttons [image].  If you click in that area on Safari, they work.

closed 2016-03-27 by LogicalChaos