donovan6000/M33-Fio #125: Bed Leveling?

droidmunkey, 2016-04-03:
How does "Calibrate bed Center Z0" work?  is there a distance sensor on the print head?

Any chance we will get the bed height calibration features?  At the very least it'd be nice to be able to print a test border to see if you need to adjust it.

https://printm3d.com/solutions/article.php?id=216

donovan6000, 2016-04-03:
Calibrate bed center Z0 is equivalent to the generic calibrate bed in M3D's official software. It runs the G-code command G30. There's an accelerometer in the extuder's housing that M3D calls their Micro Motion Sensor Chip which is used to determine where Z0 is.

M3D Fio adds an entire section to OctoPrint's controls tab that's dedicated to calibrating the bed. The option to print a test border is in that section.
[image]

You can manually change the corner offsets and bed height offset by clicking the 'Print settings' button in the general section. Those settings are equivalent to what the cat screen in M3D's official software is. However you should just be able to use the 'Run complete bed calibration' button to have your bed calibrated accurately without having to manually fine tune any settings.
[image]

droidmunkey, 2016-04-03:
Wow i must be blind I totally didn't see the test border before...  I am running the complete bed calibration now.

droidmunkey, 2016-04-03:
Does this fix the bed offsets? I have already changed them trying to manually level the bed with the 'Save Z as front left ZO' and the such.

donovan6000, 2016-04-03:
Yes, it'll fix the bed corner offsets. It does the following which overwrites the bed offsets with new values.
[image]

droidmunkey, 2016-04-03:
Yea I figured that out after I hit it, It'd be nice to be able to cancel it without changing the offsets when you get to the manual part.  Thanks again for the awesome software, you made my printer usable :-)

closed 2016-04-08 by donovan6000