donovan6000/M33-Fio #126: Add 315°C support for  V2016040401

ra100, 2016-04-09:
can you add high tempereture support to newest firmware? Thanks.

donovan6000, 2016-04-09:
People were complaining about having to flash the firmware everytime they switched between using M3D Fio and the official software due to the version number I used in the last 315°C firmware, so I M3D Fio now includes both unmodified and modified versions of the firmware.

The M3D Mod firmwares have the temperatures increased, so M3D Mod V2016040401 is what your looking for.
[image]

ra100, 2016-04-09:
Well, that's interesting, I don't have Mod version in menu, but nevermind, I just went through files, and I can flash it from .hex file.
[image]

closed 2016-04-09 by ra100

ra100, 2016-04-09:
I forgot to remove build/ dist/ and OctoPrint_M3DFio.egg-info/ folders before python setup.py install. Sorry.