donovan6000/M33-Fio #127: Backlash Calibration cylinder prints at wrong Z offset

ra100, 2016-04-09:
When I tried to print calibratin cylinder, head moved too low, seemed ike at least 1mm lower than it should and ofter stopping print I needed to recalibrate corner offsets, to be able to print test border.
I hope that relevant parts are in this log:
[log 20160409.zip](https://github.com/donovan6000/M3D-Fio/files/211495/log.20160409.zip)

ra100, 2016-04-10:
I tried it also with some model [backlash.stl.zip](https://github.com/donovan6000/M3D-Fio/files/211734/backlash.stl.zip) and result was the same, model started printing below bed (luckily I've got glass, so the nozzle tip was driving on glass, with no space between but no damage done, thanks to glass :) ), so I stopped print and when I try to print test border, it prints just fine.
Here is log from printing border [log 20160410Border.zip](https://github.com/donovan6000/M3D-Fio/files/211737/log.20160410Border.zip)
And this is model printing log 
[log 20160410Model.zip](https://github.com/donovan6000/M3D-Fio/files/211738/log.20160410Model.zip)

ra100, 2016-04-13:
Maybe I'm wrong, but it seems calibration process should be changed. And that bed height (center) doesn't affect corner offsets. So in the corners you can have head just above bed and in center much higher or lower. It's just a theory but, seems like combination of bed height compensation and material weariness. That when head is closer to center, tracks need to go higher to keep nozzle at right height.
[image]
And something like this would look head height without compensation.
[image]

donovan6000, 2016-04-19:
Sorry this took so long for me to reply to. I actually had to think about the slight change in height due to the print head's position for a few days since I never considered that before. I believe that the currently calibration process/bed compensation pre-processor already takes that height variation into account.

I agree that when the extruder moves away from the center that the print head will be slightly higher due to material weariness. However moving away from the center doesn't change the printer's internal Z value, so the printer will always think it's at the same Z value no matter how it's X/Y values change. The calibration process/bed compensation pre-processor uses each corner's relative vertical displacement from the center, and that displacement includes the height variation due to material weariness because of the printer's internal Z value doesn't change when it moves to each corner.

For example when it goes to calibrate a corner it moves to Z3, but what the printer doesn't know is that it's really at Z3 + ΔZ due to the material weariness. This results in the vertical displacement for that corner to account for that ΔZ because it didn't actually start at Z3. I also feel like the test border proves that either ΔZ is being accounted for or that ΔZ is too small to really affect a print since the test border's height does measure very uniformly everywhere when the bed is calibrated accurately.

As for the original issue in this post, I have no idea why the backlash cylinder printing about 1mm lower than it should have. The logs indicate that it never goes below Z0.479522. The test border log shows that it's at Z0.4 frequently, so it's pretty weird that the test border printed fine when it was at a lower height. I've been having some weird issues with the V2016040401 firmware where the stored Z value in EEPROM will suddenly change to some huge value, so that might be contributing to this. Can you pause the print and run an M114 command if this happens again so we can see where the printer thinks it's current Z value is?

ra100, 2016-05-01:
I haven't done with my printer anything for some time. I just tried to recalibrate, twice and result is close to the same. I tried to print another model [sticktest.gco.zip](https://github.com/donovan6000/M3D-Fio/files/244254/sticktest.gco.zip)
, and this I've got from M114 when I paused print.

closer to center> `Recv: ok  X:18.2384 Y:49.9546 E:2.0756 Z:0.8791`

closer to border: `Recv: ok  X:50.6816 Y:9.2246 E:8.7309 Z:0.2626`

Test border is fine, but anything in the center should be about 0.1 - 0.4mm higher

donovan6000, 2016-09-21:
After looking through all the log files again, I'm convinced that this is problem results from inaccuracies in the printer's Z0 calibration.

Let me know if you have any interest in further working on this issue, otherwise I'll close it in a few days.

ra100, 2016-09-21:
I think, this issue can be closed, seems that calibration works best, when I calibrate all offsets manually, starting with center Z0.

closed 2016-09-21 by ra100