donovan6000/M33-Fio #130: Brand new printer, suddenly thinks "home" is center of left side

hukuzatuna, 2016-04-15:
I had no luck with the default M3D software, so installed Octoprint on an RPi2 with M3D-Fio. I have had one successful print (the first one on the new M3D), and now when I press "home" under X/Y in the movement section, the extruder moves to the center of the left side. Recalibrating the bed orientation doesn't help. Power cycling doesn't help. Rebooting the Octoprint server doesn't help. I'm not sure this is an M3D-Fio problem or something else. Suggestions would be greatly appreciated.

donovan6000, 2016-04-16:
All M3D Fio does to home the printer is send it a G28 command, so this sounds like an issue with the printer's firmware. Try installer the M3D V2016010812 firmware from the Advanced section of OctoPrint's Control tab to see if that helps. That firmware tends to be their most reliable.

Also can you control the X/Y movement with the up, down, left, right buttons without any issues? If not then you printer might have a faulty motor.

donovan6000, 2016-05-27:
It's been awhile since this issues had a response, so I'll probably close it in a couple of days. Let me know if you still have any interest in resolving this issue.

hukuzatuna, 2016-05-27:
I'm sorry, I should have responded sooner. I upgraded to the latest version
of M3D-Fio (at Pedro and Luis' suggestion from Adafruit), upgraded
OctoPrint, upgraded the firmware, and lo and behold the printer has been
working flawlessly since then. I suspect there was a bug in an earlier
version of the firmware that I was consistently exercising. Please go ahead
and close this issue, as I have a working configuration now.

donovan6000, 2016-05-27:
Awesome! Glad to hear you got everything working.

closed 2016-05-27 by donovan6000

hukuzatuna, 2016-05-28:
Thank you for making and supporting M3D-Fio! I'm absolutely certain that if
it weren't for you and OctoPrint I would still have a bricked printer.....

Cheers,
Phil