donovan6000/M33-Fio #131: Got stuck on warming up. How do I reset everything?

mydyn96, 2016-04-18:
While getting everything set up, octoprint decided to explode when it got frozen on Warming Up... At this point I would like to uninstall both octoprint and m3d-fio and reinstall them from scratch, with none of the old stuff remaining. How do I do that? Thanks

donovan6000, 2016-04-18:
There's uninstallers available for [Windows](https://raw.githubusercontent.com/donovan6000/M3D-Fio/master/installers/Windows/uninstall.zip), [OS X](https://raw.githubusercontent.com/donovan6000/M3D-Fio/master/installers/OS%20X/uninstall.zip), and [Linux](https://raw.githubusercontent.com/donovan6000/M3D-Fio/master/installers/Linux/uninstall.zip) that uninstall Python, OctoPrint, M3D Fio, and any device drivers. After that you can use the installers to preform a fresh install.

mydyn96, 2016-04-18:
Thank you for the very quick response. The biggest issue I'm having is that when I reinstall everything and open it back up, the old stuff is still on the server and I cannot log in. I guess what I'm really asking is how to clear the old server and account.

donovan6000, 2016-04-18:
You'll need to delete OctoPrint's config.yaml and users.yaml files. They're in `%SYSTEMDRIVE%\Users\%USERNAME%\AppData\Roaming\OctoPrint` in Windows, `/Users/$USER/Library/Application Support/OctoPrint` in OS X, and `/home/$USER/.octoprint` in Linux.

mydyn96, 2016-04-18:
Alright, turns out I needed to end the current process to reset it. However, I am still having issues with the software. One is when I run the complete bed calibration, it asks about the external bed, I click no, and it moves to the correct position and then does nothing; it just freezes. Another issue is when I set a temperature and click the button to warm it up. It says warming up... and then immediately says done, but it does not actually warm up. If I try to click the warm button again it just stays on Warming Up... forever. Sorry to dump all these issues on you and thanks for the help.

donovan6000, 2016-04-19:
Try using [M3D Fio's devel branch](https://github.com/donovan6000/M3D-Fio/tree/devel) to see if those problems are fixed now. You can install it with OctoPrint's builtin plugin manager by installing a plugin from this URL.
```
https://github.com/donovan6000/M3D-Fio/archive/devel.zip
```

mydyn96, 2016-04-24:
Thanks a ton! The printer has never worked this well before.

closed 2016-04-25 by donovan6000