donovan6000/M33-Fio #135: Super slow printing and Print Time Estimates off

Fost325, 2016-04-24:
When Printing with M3D Fio I have been getting super slow prints.
Also my estimated times are off completely. 

This file has been printing for 23 hours now. It still says it has 4 more hours. It was only supposed to take around 17.
[image]

I realize it is only a $400 dollar printer surely printing a 2 inch figure with slightly above average quality and medium infill should not take 27 hours to print.

Im completely new to this. Only had this printer for a few days. Im accustomed to a Dimension 1200es which does all the work for you. I purchased this for home use to learn the "backside of printing".

Thanks for any advice.

donovan6000, 2016-04-25:
OctoPrint's API doesn't have any way for plugins to change the displayed print times, so there's nothing M3D Fio can do about this.

By default G-code being printed gets pre-processed by M3D Fio right before being sent to the printer to compensate for things like backlash and bed leveling which can easily triple the number of command being sent, so OctoPrint's isn't aware of all the commands actually being sent to the printer which can contribute to the estimated times being inaccurate.

You can disable the 'Pre-process on the fly' option to have all the G-code pre-processed when it gets uploaded/sliced to OctoPrint which will make the estimated times more accurate since OctoPrint will know all the commands that will be sent to the printer. However those pre-processed actions will be hardcoded into the upload/sliced G-code, so you'll need to reupload/reslice any files if you change the print settings and want them to be applied.

The Micro 3D is a pretty slow printer. They claim it can move up to 60mm/s, but they don't tell you that it can only reach that speed when it's only moving the X/Y motors. The printer's firmware makes it so that all X/Y/Z/E movements in a single command end at the same time, so it'll slow down the faster X/Y motors to bring them down to the Z/E motors speeds when it's changing Z or E.

donovan6000, 2016-05-27:
It's been awhile since this issues had a response, so I'll probably close it in a couple of days. Let me know if you still have any interest in resolving this issue.

closed 2016-06-01 by donovan6000