donovan6000/M33-Fio #136: Can't Connect to Printer

jaaron, 2016-04-29:
I'm trying to get Octoprint + M3D-Fio to connect to my M3D and it's not working well at all.

I'm running Octoprint (git rev 692166f06) with M3D Fio 0.32 in a Xubuntu 16.04 VM running on VirtualBox on a Mac. I have USB filter rules set up to pass the M3D to the VM (Vendor 03eb Product 2404 Revision 0100). It takes two rules because the M3D seems to be initially recognized as an Atmel Corp IOUSBHostDevice then after some initialization as an M3D Printer.

I started down this path due to frustration with the instability and poor print quality I was getting from M3D's official Mac application.

The first time I attempted to connect I was notified that there was a firmware update, from I believe 4/1/2016 (maybe I should have been more suspicious of an April Fools firmware update, but given that my printer has been doing poorly, a firmware update seemed reasonable). This seemed to work and then the printer rebooted (well, the light blinked which I assume was it rebooting). After applying this update, the official app stopped recognizing the printer entirely.

Back to OctoPrint, clicking connect in the OctoPrint gui generates the following log:

> Changing monitoring state from 'Offline' to 'Detecting serial port'
> Connecting to: /dev/ttyACM2
> Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
> Connected to: Serial<id=0x7f563b384090, open=True>(port='/dev/ttyACM2', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
> Starting baud rate detection
> Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
> Trying baudrate: 250000
> Recv: e1
> Changing monitoring state from 'Detecting baudrate' to 'Closed'
> Send: N0 M110 N0*125
> Connection closed, closing down monitor

The light on the M3D blinks once (rebooting?).

After this the printer is assigned a new device file (e.g., if it was /dev/ttyACM0 it becomes /dev/ttyACM1). I first tried adding the suggested udev rule:

> ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2404",  GROUP="users", MODE="0666", ENV{ID_MM_DEVICE_IGNORE}="1"

But this didn't seem to help (I had already added myself to the dialout group that is the default owner of the device apparently).

I modified the rule to also set NAME:

> ACTION=="add", SUBSYSTEMS=="usb", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="2404", NAME="ttyACM0", GROUP="users", MODE="0666", ENV{ID_MM_DEVICE_IGNORE}="1"

to try to make sure it had a consistent device file name, but this didn't have the desired effect (my udev-fu is weak).

Connecting screen to the tty prints "wait" repeatedly and "e1" every time I type a key (different baudrates don't seem to make a difference).

I'm pretty much out of ideas. Any help appreciated!

jaaron, 2016-04-30:
Update: I was able to use the m3d-linux command line tool to revert the firmware to a previous version that the official software was able to recognize as invalid and replace with a functional version. I can now use the official software, but OctoPrint continues to fail to connect.

donovan6000, 2016-04-30:
Make sure your have the port and baudrate both to set to auto. The printer will jump between different /dev/ttyACM ports when it switches between bootloader and firmware mode, and M3D Fio is able to accommodate for the that when the port is set to auto. The light will blink when the printer is switching modes since all the pins are low when the microcontroller starts up and get set to high by the bootloader/firmware. M3D Fio will also default to use the 115200 baudrate when the baudrate is set to auto.

And the V2016040401 firmware isn't an April fools joke. It's the firmware that comes with M3D's newest software V1.4.2.3.

Do you receive any errors like this one when you try to connect?
[image]

Also there's an [installer for OS X](https://github.com/donovan6000/M3D-Fio/blob/master/installers/OS%20X/install.sh) that installs Python, OctoPrint, and M3D Fio. Not sure if you have any particular reason of using OctoPrint through Xubuntu via VirtualBox when you could be running it directly on OS X.

jaaron, 2016-04-30:
Thanks!

I have port and baudrate both set to Auto.

Yes, I get that error. 

I didn't really think the firmware was a joke, I was making a joke that may have fallen flat. I can't find the 1.4.2.3 software on the M3D site. Where do you get it? I'm at https://www.dropbox.com/sh/maf6kv4c80fyu0b/AAD1b453nfBaPgLRSTDemknVa?dl=0#/ (linked from the printm3d support page) and only see 20160129 for windows and 20160126-beta for Mac.

I'm doing the linux VM because my plan is to move to OctoPi and doing it in a VM seemed like the easiest way to experiment without polluting my main workstation.

jaaron, 2016-04-30:
I went ahead and installed directly on my Mac and it's working happily. I blame vbox's usb passthru. Thanks for the help.

closed 2016-04-30 by jaaron

donovan6000, 2016-04-30:
Awesome! There's also an [uninstaller for OS X](https://github.com/donovan6000/M3D-Fio/blob/master/installers/OS%20X/uninstall.sh) so that you can remove it from you workstation after you get OctoPi set up.