donovan6000/M33-Fio #141: Low Power, Eventual Reboot

Alaric323, 2016-05-23:
Greetings devs,

I seem to have an issue with M3D-Fio (or maybe OctoPrint, but I'm not sure, as I've just started using the program today), in which when I try to print something through Octoprint, the M3D logo in the front dims significantly.  The difference is indicated in the photos attached. 
 
[image]
[image]

This is clearly to do with a lack of power, as after a few minutes of finding the home location and beginning to print, the extruder slows to a halt, the printer shuts down, and reboots mid-print.

I used the original M3D software to try and print after discovering this problem, and the original software prints without issue (although there is an incredibly slight flickering in the logo.)  

The original software version is 20160506, with printer firmware version unmodded 2016040401.
Octoprint is version 1.2.11 with M3D-Fio 0.33, and with Octoprint I tried the same printer firmware as above, as well as the iMe v0.07 firmware.  Both firmwares gave the issue.

I didn't seem to find a repeat issue when looking through past ones on github, but my apologies if I overlooked it.

Thanks in advance for the support!

donovan6000, 2016-05-23:
M3D Fio has this setting enabled by default which sets the LED's brightness to 1% when a print starts and flashes it several times when a print finishes. This is accomplished using the M420 G-code command, so this behavior is intentional and not a power related issue.
[image]
I'm not really sure why your printer is rebooting mid-print though. If your using a single board computer, like a Raspberry Pi, then you might want to try using a externally powered USB hub in case it's the single board computer that's having trouble maintaining the connection with the printer.

Alaric323, 2016-05-23:
I see, my apologies for mistaking this as an issue.  At least now I know that's not a big concern since it can be disabled.

As for the printer stopping mid-print, as long as its not a power issue, I suppose it just requires further tweaking of the settings.  I'm using a Dell Optiplex 755 as a dedicated print server, so its definitely not an issue about USB connection.  Thanks for the quick response.

closed 2016-05-23 by Alaric323