donovan6000/M33-Fio #143: Idea: Add support for Slic3r

ra100, 2016-05-27:
I've been thinking if it would be possible to support https://github.com/OctoPrint/OctoPrint-Slic3r. And what will I need to accomplish that. I'm not asking you @donovan6000 to do that, I'm asking if you could point me in right direction, what needs to be done.
I get that I'll need to somehow convert Cura printing profiles to Slic3r, at least values that are similar. Make/find script that would make it automatically.
And lot of other stuff I have no idea of :)

Why I'd like to try that? I like idea of hexagonal infill. And that's it.

donovan6000, 2016-05-27:
It looks like there's also a [pull request](https://github.com/OctoPrint/OctoPrint-Slic3r/pull/3) on the Slic3r plugin from a couple of days ago that claims to have working on the newest Slic3r V1.2.9, so give that a try in M3D Fio and let me know if anything related to M3D Fio doesn't work like it should.

I did designed M3D Fio while knowing that people wouldn't always be using OctoPrint's Cura Engine plugin to slice, so I made everything specific to Cura not happen if the Cura wasn't being used to slice. So the [slicer profile editor will only show the advanced section](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/static/js/m3dfio.js#L5155-L5167) and the [model editor wont display a raft/brim/skirt](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/static/js/m3dfio.js#L955-L1019) if the users isn't using Cura Engine to slice because those things only know to display those with respect to Cura's profile parameters.

Fortunately the [Slic3r plugin convertToEngine function](https://github.com/OctoPrint/OctoPrint-Slic3r/blob/master/octoprint_slic3r/profile.py#L334-L351) shows that it's only using one type of syntax, so everything about M3D Fio's python code should already work with. The only thing we'd have to change there is installing Slic3r slicer profiles if we want to provide them. I can't find much information on converting Cura profiles to Slic3r profiles, so that's something you might have to develop or create your own profiles from scratch. I was able to find a [Slic3r profile tor the M3D](http://www.thingiverse.com/thing:877137) that was made about a year ago.

So most of the work is going to be changing M3D Fio's JavaScript code to work with [Slic3r slicer profile parameters](https://github.com/donovan6000/M3D-Fio/blob/master/octoprint_m3dfio/__init__.py#L1058-L1236).

ra100, 2016-05-28:
Thanks, I'm working now on identifying what keys in cura correspond to what keys in slic3r (with some modifications). And I hope, I'll be able to create python script which converst cura to slic3r profiles.

ra100, 2016-05-30:
Seems like it will take longer than I expected. I tried to create new profile based on profiles for cura. It slices, but I'm gettig warning that Could not print the file. The dimensions of the model go outside the bounds of the printer. And when I chose to ignore that warning. Head moved to FL and after heating nozzle it moved ~4cm up and just started to extrude, slowly and octoprint disconnected.
I'm atatching gcode, serial log and profile.
Maybe problem is, that added to this profile start and end gcode from your profiles.
[test.zip](https://github.com/donovan6000/M3D-Fio/files/289955/test.zip)

Javierma, 2016-05-31:
Sorry, I totally forgot to specify that tests were done under a Raspberry Pi so it may not work under other Operating Systems. If that is the case, you may want to execute the following script (it compares files with gcode instructions) to make sure that it works as expected (I tried with several files and worked, but under Raspberry Pi as I said):

file_plugin=name_of_file_generated_using_plugin
file_slic3r=name_of_file_generated_using_Slic3r

gco_content=[]
gcode_content=[]

with open(file_plugin) as f:
	for line in f:
		gco_content.append(line)

with open(file_slic3r) as f2:
	for line in f2:
		gcode_content.append(line)

for content in gco_content:
	try:
		gcode_content.remove(content)
	except ValueError:
		continue
		
print "\nContent after comparison\n"
for content in gcode_content:
	print content

print "\nTotal different lines: "+str(len(gcode_content))

donovan6000, 2016-05-31:
Slic3r uses [] instead of {} for all their variable parameters, so you'll need to change all the settings like `F{travel_speed}` to `F[travel_speed]` in all the _gcode settings, like start_gcode and end_gcode, in order for Slic3r to replace those variables with their settings when it slices a file. This is what the resulting profile would look like.
[m3d_slic3r_2.zip](https://github.com/donovan6000/M3D-Fio/files/290720/m3d_slic3r_2.zip)

However the `retraction_dual_amount` setting doesn't exists, so this profile still wont slice a printable file  since it'll have [retraction_dual_amount] in that sliced file. 

I didn't use the start_gcode or end_gcode equivalents in the Cura profiles at all since M3D Fio's preparation pre-processor seems to take care of everything pretty well, so maybe we should just set all the _gcode settings to nothing in the Slic3r profiles?

ra100, 2016-05-31:
Thanks, I'll try it again later today. And I'm not 100% sure if print volume is set correctly, and some slic3r values are just guesses, in some cases I couldn't find documentation (or what I wanted to find). And maybe I should just remove some keys/values from profile and let them be omitted or set to default by slic3r plugin.
And I'm running OctoPrint on RPi2. Just for the record.

ra100, 2016-05-31:
Well, I have first working profile. I needed to change gcode flavor value (makerware is using A as extruder axis, not E). I still need to figure out how print_center works and if it works at all, when I tried to set [54,54] and print just 5mm cube, it was printed in BR corner. Maybe it would be for the best, to set center to [0,0] and in M3D-Fio choose center print (this way I'm able to print larger object without warnings). Now, I think, can move to the next phase, create automatic conversion from cura to slic3r and then add gui for profile editor.

[m3d_slic3r_3.zip](https://github.com/donovan6000/M3D-Fio/files/291807/m3d_slic3r_3.zip)

donovan6000, 2016-05-31:
Awesome! [I added TODOs](https://github.com/donovan6000/M3D-Fio/commit/870abadabbf78f3b5f43ba5777d0e1dc53ea9329) into all the places in all the files that we'll need to change, __init__.py and m3dfio.js. I'll try to get M3D Fio's automatic Slic3r path detection and profile importing done today.

donovan6000, 2016-05-31:
All that's left is the m3dfio.js changes now, which will be easy once we know how to convert between Cura and Slic3r settings. I wasn't able to position the model at all, even when changing the printer_center and bed_size parameters, so I guess this is just a limitation of the Slic3r plugin.

donovan6000, 2016-05-31:
Running Slic3r manually has shown that it doesn't use the print_center, bed_size, or bed_shape settings when deciding the model's position. The only way I was able to change the position was to pass the --print-center command line argument to it, so we'll have to change the Slic3r plugin to also do that.

ra100, 2016-06-01:
Great job.
I've just created cura->slic3r [conversion table](https://github.com/ra100/curaToSlic3rProfile/blob/master/Conversion.md), I'm still not 100% sure about all values, but I think, that most important ones I got right.

donovan6000, 2016-06-14:
Everything about M3D Fio's support for OctoPrint's Slic3r plugin is all working now. Thanks for all the help :)

Seems like the only thing left is to wait for OctoPrint's Slic3r plugin to mature. In the meantime I'll tell people to use [your branch](https://github.com/ra100/OctoPrint-Slic3r) if they want to try out M3D Fio's Slic3r support.

closed 2016-06-14 by donovan6000

ra100, 2016-06-14:
Probably you can tell them to use [Javierma/OctoPrint-Slic3r](https://github.com/Javierma/OctoPrint-Slic3r), my changes are already merged in devel branch. Either way, I intend to keep it up to date with Javierma's changes.