donovan6000/M33-Fio #144: Issue installing on octopi 1.3 jesse

mabawsa, 2016-05-30:
Hi,

I am trying to install M3D-Fio on a pi fresh install of octopi 1.3. I am getting the following error repeated for every pip command:

You are using pip version 8.1.1, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Can not perform a '--user' install. User site-packages are not visible in this virtualenv.

Please advise.

ra100, 2016-05-30:
Can you try to install it as global package with `sudo`? Or install it from repository

```bash
git clone https://github.com/donovan6000/M3D-Fio.git
cd M3D-Fio
python setup.py install
```

mabawsa, 2016-05-30:
pi@octopi:~/M3D-Fio $ python setup.py install
Traceback (most recent call last):
  File "setup.py", line 2, in <module>
    import setuptools
ImportError: No module named setuptools
pi@octopi:~/M3D-Fio $ python setup.py install

is the response.

I can install using the GUI for plugins but I cannot then connect to the printer.

ra100, 2016-05-30:
Seems like you'll need to install setuptools first

```bash
pip install setuptools
```
(or with sudo)

and then
```bash
python setup.py install
```

or just try

```bash
sudo pip install https://github.com/donovan6000/M3D-Fio/archive/master.zip
```

another thing, I don't know if you are using virtualenv, if you have octoprint installed in virtual environment, I can't help you much, I can just guess. Or I'll need to try to install octoprint that way myself.

What is your system? How you installed OctoPrint?

mabawsa, 2016-05-30:
a raspberry pi, using octopi.
None of the pip commands work, re dd'ing an image atm.

ra100, 2016-05-30:
If it doesn't help. And installation via GUI doesn't work. Try this:

```bash
wget https://raw.githubusercontent.com/donovan6000/M3D-Fio/master/installers/Linux/install.zip
unzip install.zip
sudo sh ./install.sh
```

This will manually do what GUI install should. Important thing is that you have m3d rules in `/etc/udev/rules.d/`.

mabawsa, 2016-05-30:
it was a bad octopi install. I reinstalled and used the gui to install the plugin.
I can connect to the printer.
Anything else I need to do?

ra100, 2016-05-30:
If you can connect to printer and everything works fine, there is no need to do anything else. Just enjoy and print :) And you can close this issue.

closed 2016-05-30 by mabawsa