donovan6000/M33-Fio #147: ImportError: cannot import name IncompleteRead when using `install.sh`

johnwebbcole, 2016-06-04:
I'm running into an issue installing on a RPi2 with the latest jesse-lite.  Using

```
wget https://raw.githubusercontent.com/donovan6000/M3D-Fio/master/installers/Linux/install.zip
unzip install.zip -x uninstall.sh
sudo sh ./install.sh
```

I end up looping on `ImportError: cannot import name IncompleteRead`

```
Get:5 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9,091 kB in 3min 12s (47.2 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version.
libjpeg-dev is already the newest version.
libyaml-dev is already the newest version.
python is already the newest version.
python-dev is already the newest version.
python-pip is already the newest version.
zlib1g-dev is already the newest version.
python-pygame is already the newest version.
The following package was automatically installed and is no longer required:
  device-tree-compiler
Use 'apt-get autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 7, in <module>
    from pip import main
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 7, in <module>
    from pip import main
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python2.7/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name IncompleteRead
octoprint: no process found
--2016-06-04 13:10:12--  https://github.com/foosel/OctoPrint/archive/master.zip
Resolving github.com (github.com)... 192.30.252.129
Connecting to github.com (github.com)|192.30.252.129|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/foosel/OctoPrint/zip/master [following]
--2016-06-04 13:10:13--  https://codeload.github.com/foosel/OctoPrint/zip/master
Resolving codeload.github.com (codeload.github.com)... 192.30.252.161
Connecting to codeload.github.com (codeload.github.com)|192.30.252.161|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘master.zip.2’
```

Searching on the error, seems like a lot of people are running into it:

http://stackoverflow.com/questions/27341064/how-do-i-fix-importerror-cannot-import-name-incompleteread

But none of the offered solutions have worked for installing M3D-Fio.  

Any suggestions for getting around the error?  The install.sh loops over and over on this error.

donovan6000, 2016-06-05:
It seems like this is caused by an old version of requests being loaded by Python instead of the newer version. This happens since Python is searching `/usr/local/lib/python2.7/dist-packages` before it searches `/usr/lib/python2.7/dist-packages` when importing a module, so it's finding the older version first. Running the following caused me to get the same error as you.
```
python -c "from requests.compat import IncompleteRead"
```
And I was able to fix it by just removing the old version of requests.
```
sudo rm -rf /usr/local/lib/python2.7/dist-packages/requests*
```

donovan6000, 2016-09-21:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2016-09-23 by donovan6000