donovan6000/M33-Fio #149: can't upgrade fio from .31 to .34

stephenci, 2016-06-06:
while trying to upgrade to 0.34 I encountered the following error. this is with firmware M3D 2016040401
---------------------------------------------------------------------------
Installing plugin from https://github.com/donovan6000/M3D-Fio/archive/V0.34.zip...
C:\python\Scripts\pip.exe install https://github.com/donovan6000/M3D-Fio/archive/V0.34.zip --user
Collecting https://github.com/donovan6000/M3D-Fio/archive/V0.34.zip
Downloading https://github.com/donovan6000/M3D-Fio/archive/V0.34.zip
Installing collected packages: OctoPrint-M3DFio
Found existing installation: OctoPrint-M3DFio 0.31
Uninstalling OctoPrint-M3DFio-0.31:
Successfully uninstalled OctoPrint-M3DFio-0.31
Running setup.py install for OctoPrint-M3DFio
Exception:
Traceback (most recent call last):
File "c:\python\lib\site-packages\pip\basecommand.py", line 211, in main
status = self.run(options, args)
File "c:\python\lib\site-packages\pip\commands\install.py", line 311, in run
root=options.root_path,
File "c:\python\lib\site-packages\pip\req\req_set.py", line 657, in install
requirement.commit_uninstall()
File "c:\python\lib\site-packages\pip\req\req_install.py", line 729, in commit_uninstall
self.uninstalled.commit()
File "c:\python\lib\site-packages\pip\req\req_uninstall.py", line 152, in commit
rmtree(self.save_dir)
File "c:\python\lib\site-packages\pip\_vendor\retrying.py", line 49, in wrapped_f
return Retrying(*dargs, **dkw).call(f, *args, **kw)
File "c:\python\lib\site-packages\pip\_vendor\retrying.py", line 212, in call
raise attempt.get()
File "c:\python\lib\site-packages\pip\_vendor\retrying.py", line 247, in get
six.reraise(self.value[0], self.value[1], self.value[2])
File "c:\python\lib\site-packages\pip\_vendor\retrying.py", line 200, in call
attempt = Attempt(fn(*args, **kwargs), attempt_number, False)
File "c:\python\lib\site-packages\pip\utils\__init__.py", line 90, in rmtree
onerror=rmtree_errorhandler)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 247, in rmtree
rmtree(fullname, ignore_errors, onerror)
File "c:\python\lib\shutil.py", line 252, in rmtree
onerror(os.remove, fullname, sys.exc_info())
File "c:\python\lib\site-packages\pip\utils\__init__.py", line 102, in rmtree_errorhandler
func(path)
WindowsError: [Error 5] Access is denied: 'c:\\users\\stephen.cia\\appdata\\local\\temp\\pip-f3umjo-uninstall\\users\\stephen.cia\\appdata\\roaming\\python\\python27\\site-packages\\octoprint_m3dfio\\static\\libraries\\preprocessor_i386.dll'
You are using pip version 7.1.2, however version 8.1.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
Done!
-----------------------------------------------------------------------------------

donovan6000, 2016-06-06:
It's trying to uninstall V0.31 before installing V0.34, but it can't uninstall the old version since it's currently using pre-processors shared library which makes it so that it can't delete that file while V0.31 is running. The only way I can think of fixing this is to load the shared library when M3D Fio needs it and unload it when it's done with it, so this is something I'll have to fix in M3D Fio.

As a temporary workaround you can kill OctoPrint manually which will kill M3D Fio so that you can install the update.
```
taskkill /IM pythonw.exe /T
C:\python\Scripts\pip.exe install https://github.com/donovan6000/M3D-Fio/archive/V0.34.zip --user
START "" "%SYSTEMDRIVE%\python\pythonw.exe" -c "import octoprint;octoprint.main()"
```

stephenci, 2016-06-06:
hmm without doing what you suggested, I restart my computer and go back to fio to check the version it is now showing 0.34 latest.... not sure if this means it is installed properly? but I did see Muele's calibration button which was not in 0.31 version. So I think it install ok...

donovan6000, 2016-06-07:
I guess it updated to V0.34 somehow. The next version of M3D Fio makes it so that the shared library gets unloaded when M3D Fio is done using it, so that should prevent anyone from experiencing this issue in the future.

closed 2016-06-14 by donovan6000