donovan6000/M33-Fio #150: Error in unloadSharedLibrary

ra100, 2016-06-07:
I'm getting this ugly error on Linux, before print, when using shared library.

```log
2016-06-07 20:00:30,353 - octoprint.plugin - ERROR - Error while calling plugin m3dfio
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.0.dev1071+gdb66e4a-py2.7.egg/octoprint/plugin/__init__.py", line 242, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.34.2016060604-py2.7.egg/octoprint_m3dfio/__init__.py", line 4554, in on_event
    if self.loadSharedLibrary() :
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.34.2016060604-py2.7.egg/octoprint_m3dfio/__init__.py", line 933, in loadSharedLibrary
    self.unloadSharedLibrary()
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint_M3DFio-0.34.2016060604-py2.7.egg/octoprint_m3dfio/__init__.py", line 1044, in unloadSharedLibrary
    _ctypes.dlclose(self.sharedLibrary._handle)
OSError: P�rP�r�r�r`�r`�rh�rh�rp�rp�rx�rx�r��r��r��r��r��r��r��r��r��r��r��r��r��r��r�<�r�<�r��r��r��r��r�B�r�B�r�0�r�0�r��r��r��r��rp��rp��r��r��r: shared object not open
```

and withou shared library
```log
Fatal Python error: (pygame parachute) Segmentation Fault
fish: “octoprint serve” terminated by signal SIGABRT (Abort)
```

donovan6000, 2016-06-07:
Alright, it should be fixed now. [Windows has trouble updating M3D Fio when it has the shared library loaded](https://github.com/donovan6000/M3D-Fio/issues/149), so I changed it so that it'll only load the shared library when it needs it and unload it when it's done.

ra100, 2016-06-08:
I confirm, it's fixed. Thanks.

closed 2016-06-08 by ra100