donovan6000/M33-Fio #151: Z offset

qwerty8224, 2016-06-08:
It is possible to add such an option?

donovan6000, 2016-06-08:
If I'm understanding you correctly, then I'd say the bed height offset setting is what your looking for. M3D Fio applies that offset to all Z value when printing.

qwerty8224, 2016-06-08:
I thought it was something quite different))) Thank you. But it is as it does not always work. Could you check it. I use M3D fio 0.34

donovan6000, 2016-06-08:
I just tried it with both the shared library pre-processor and Python pre-processor, and they both produced the expected results. As you can see, every Z value in the `calibration_cube 5mm.gco` file is 5mm higher than the Z values in the `calibration_cube 0mm.gco` since the bed height offset was set to 5mm when slicing the `calibration_cube 5mm.gco` file and 0 when slicing the `calibration_cube 0mm.gco` file.

[bed height offset.zip](https://github.com/donovan6000/M3D-Fio/files/305798/bed.height.offset.zip)

closed 2016-06-09 by qwerty8224