donovan6000/M33-Fio #152: Loading external filament

endotronic, 2016-06-09:
Forgive me if this isn't the best place to ask a question, but I don't know where else to ask. M3D Fio works really well for me, and has made the M3D Micro a much better printer! Thank you so much! I would like to be able to load external filament through Octoprint / M3D Fio, but I have only been able to load internal filament. Should the load button in the controls pane work for either? Is there a different way to load external filament?

Thanks!

donovan6000, 2016-06-09:
Yes, the load button works for loading both internal and external filament. Loading internally and externally both feed the filament to the same place and use the same motor, so there only needs to be one button to handle both.

M3D's official software saves the filaments type, color, and location in the printer's EEPROM, which is why it asks you so many questions whenever you change the filament. I don't do the same thing in M3D Fio since those settings never get used by the printer's firmware, so there's really no point to saving them in the printer's EEPROM.

endotronic, 2016-06-09:
OK thanks, I must have just not gotten the filament all the way in there. I'll mark this issue closed.

closed 2016-06-09 by endotronic