donovan6000/M33-Fio #154: Add the ability to change the M3D LED brightness

w1ll1am23, 2016-06-13:
It would be cool to have the ability to turn off the LED light on the M3D. If the issue the commands listed on M3D's site https://printm3d.com/solutions/article.php?id=227 I can turn it off with OctoPrints terminal by issuing M420 T0. Would it be possible to add a menu option (maybe with a slider) to adjust the brightness?

donovan6000, 2016-06-14:
There's now buttons in M3D Fio V0.35 that turn the LED on and off.
[image]
Keep in mind that by default the Preparation pre-processor already dims the LED when a print starts and flashes it a few times when a print is done, so you can disable M3D Fio's 'Change LED brightness' setting if you don't want that behavior.
[image]

w1ll1am23, 2016-06-14:
Nice, thanks. I wasn't getting updates, and now that I think about it I think there was a comment about that awhile ago, I was stuck on .32. Just upgraded to .35 and that worked! Thanks.

closed 2016-06-14 by donovan6000