donovan6000/M33-Fio #156: Empirically !!!

qwerty8224, 2016-06-14:
I spent a lot of tests and experiments showed that the Travell_speed  is the culprit .... shear axis. Please correct profiles travel_speed 20 ->15 The speed will not lose a lot)))) but get quality prints

qwerty8224, 2016-06-14:
It is advisable to test on many machines because it helped me.

donovan6000, 2016-06-14:
The [travel_speed in M3D Fio's Cura profiles is the same as the moveSpeed](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/cura/profile.py#L863) used in the Cura Engine parameters generated by M3D's official software. M3D knows their hardware better than anyone, which is why all of [M3D Fio's Cura profiles are just converted versions of the Cura Engine parameters generated by M3D's official software](https://github.com/donovan6000/M3D-Fio/blob/master/tools/EngineToProfile.py).

Each printer is going to have different limitations based on the wear on the motors and the environment that the printer is in. So some people's printers can handle a travel_speed of 20mm/s. You can adjust your printer's X and Y speed limits if your printer isn't capable of handling the 20mm/s travel_speed. 20mm/s is 1200mm/min and 15mm/s is 900mm/min.

qwerty8224, 2016-06-15:
Thank you) will look for their own personal settings. I thought that they were all the same.

closed 2016-06-16 by qwerty8224