donovan6000/M33-Fio #160: Cannot connect to my printer

Fullmono, 2016-06-19:
Hi, 

I can't connect to my printer, don't understand why: 

> Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6f5d0370, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 250000
Changing monitoring state from 'Detecting baudrate' to 'Closed'
Connection closed, closing down monitor
Send: M110

My configuration:
I'm running octoprint v1.2.9 on a DS215J synology NAS. Saddly, I can't update octroprint.

Thank you for your support

donovan6000, 2016-06-20:
It sounds like something caused M3D Fio to crash which is causing OctoPrint to use its native connection routine instead of M3D Fio's. Can you upload the octoprint.log file after it fails to connect? You can get it by downloading it from the Log section in OctoPrint's settings.
[image]

Fullmono, 2016-06-22:
Here your are. 
Nothing appear on the log file (I've changed logging level to debug for all plugins)
[octoprint_log_for_m3Dfio.txt](https://github.com/donovan6000/M3D-Fio/files/327811/octoprint_log_for_m3Dfio.txt)

Fullmono, 2016-07-20:
up?

donovan6000, 2016-07-22:
Not sure why it's not connecting. The log doesn't show very much. It doesn't even show that a client connected to OctoPrint. Try using the latest M33 Fio version, V1.2, and let me know if the issue persists.

Fullmono, 2016-08-23:
Sorry for the delay...
The issue persists. Any idea to solve it?
Thank you for your great job!

donovan6000, 2016-08-23:
I'm still not sure what the problem is. Make sure you have M3D Fio uninstalled and M33 Fio installed. Do you have read and write access to the printer's serial port? Does OctoPrint show any pop up messages that have errors when it fails to connect to the printer? Does your OS have any programs installed that are trying to communicate with the printer at the same time as OctoPrint, such as [ModemManager](https://www.freedesktop.org/wiki/Software/ModemManager/)?

dieideeistgut, 2016-08-23:
Re-Installing everything from OctoPrint to M33 Fio fixed this issue for me like 2 weeks ago. Taking a peek in the installer-script and following the precise order made this connection problem disappear.

donovan6000, 2016-09-21:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2016-09-23 by donovan6000