donovan6000/M33-Fio #163: M3D Profile auto-default, won't let me set any others .

SechyMonkey, 2016-06-22:
I had an existing installation of OctoPrint on Raspberry Pi Jessie. Using the Plugin Manager, I installed M3D-Fio version 0.35. After installing, it set the new profile that was created for the M3D Micro to default, and it will not allow me to set my Taz 5 profile that was already there back to default.

donovan6000, 2016-07-22:
Make sure you have the newest version of M33 Fio installed, V1.2, and that you have the 'Not using a Micro 3D printer' option enable. That should allow you to set the default profile and use other printers while M33 Fio is installed.
[image]

donovan6000, 2016-09-21:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2016-09-21 by SechyMonkey