donovan6000/M33-Fio #167: Turn the heater off for calibration

mweisshaupt1988, 2016-07-07:
Hi,

I have ruined my BuildTak sticker by running a calibration after a failed print but obviously the heater was still on and the nozzle burned through the printbed while calibrating. Is there a way to turn off the heater automatically before calibration or show the user a warning message?

If not that would be a great addition because It will be just a matter of time until I forget to turn off the heater again if something does not work :D

BTW: Thank you all your hard work ;-)

donovan6000, 2016-07-07:
The 'Calibrate bed center Z0', 'Calibrate bed orientation', and 'Run complete bed calibration' buttons all already turn off the heater before running the calibration G-codes. They follow the same procedure as M3D's official software where they [heat up the heater to 150 degrees, turn off the heater, turn off the fan, then calibrate](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/static/js/m33fio.js#L8560-L8567).

mweisshaupt1988, 2016-07-08:
Hm, interesting. I'll take a close look at the printer while calibrating in the future and I'll save the logs if this happens again. The code looks good to me so this shouldn't happen at all. It was propably my fault by doing something wrong :D

closed 2016-09-21 by donovan6000