donovan6000/M33-Fio #170: After Update got an error and Printer won't connect

ktapper, 2016-07-22:
This is the error the log reports:

Traceback (most recent call last):
  File "C:\Users\ktapper\AppData\Roaming\Python\Python27\site-packages\octoprint-1.2.13-py2.7.egg\octoprint\plugin\__init__.py", line 204, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "C:\Users\ktapper\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4567, in on_event
    self.setFileLocations()
  File "C:\Users\ktapper\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 4427, in setFileLocations
    self.convertCuraToProfile(profileLocation + profile, profileDestination + profileName, profileName, profileIdentifier, "Imported by M33 Fio on " + time.strftime("%Y-%m-%d %H:%M"))
  File "C:\Users\ktapper\AppData\Roaming\Python\Python27\site-packages\octoprint_m33fio\__init__.py", line 1195, in convertCuraToProfile
    os.remove(curaProfile)
WindowsError: [Error 32] The process cannot access the file because it is being used by another process: 'c:\\users\\ktapper\\appdata\\local\\temp\\tmpftsgih'
2016-07-21 21:39:59,957 - octoprint.server.util.sockjs - INFO - Client connection closed: 127.0.0.1
WindowsError: [Error 32] The process cannot access the file because it is being used by another process: 'c:\\users\\ktapper\\appdata\\local\\temp\\tmpftsgih'

Please help

donovan6000, 2016-07-22:
Should be fixed now in V1.2

ktapper, 2016-07-22:
Excellent let me know when released and I will check.

donovan6000, 2016-07-22:
[Already released](https://github.com/donovan6000/M33-Fio/releases)

ktapper, 2016-07-22:
Works.  Thanks for the Fix!!!

closed 2016-07-22 by ktapper