donovan6000/M33-Fio #172: Can't Connect to Printer

Alaric323, 2016-07-29:
Greetings Donovan,

After updating to the latest version of OctoPrint (1.2.14) and M33 Fio (1.2), I can no longer connect to the printer.  The log given when attempting to connect is as follows:

Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: ok 0
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok 0
Send: M21
Recv: ok
Recv: wait
Send: N0 M110 N0*125
Recv: ok
Send: M117
Unexpected error while writing to serial port: TypeError: 'unsupported operand type(s) for +=: 'int' and 'NoneType'' @ comm.py:_do_send_without_checksum:2039
Recv: PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.06 MACHINE_TYPE:Micro_3D SERIAL_NUMBER:BK16041502050017
Connection closed, closing down monitor
Changing monitoring state from 'Operational' to 'Offline: TypeError: 'unsupported operand type(s) for +=: 'int' and 'NoneType'' @ comm.py:_do_send_without_checksum:2039'
Closing down send loop

The printer was running with the latest iMe Firmware (00.00.01.06), but just to make sure it wasn't the firmware, I used M33 Manager to install the official M3D firmware, version 2016040401.  Same error occurred.  

Any advice?

donovan6000, 2016-07-29:
I didn't even notice a new version of OctoPrint got released. Let me try it out and see what's causing this.

donovan6000, 2016-07-29:
It's all fixed now in [M33 Fio V1.3](https://github.com/donovan6000/M33-Fio/releases).

Alaric323, 2016-07-29:
Thanks for the speedy resolution.  Once the update is pushed to OctoPrint's plugin repository, I'll get back to you on how it performs.

Alaric323, 2016-07-29:
The connection problem is indeed fixed, however while in the middle of an automatic full calibration, when adjusting the levels of the corners, it seems to lose connection to the printer.  The only fix is a reboot of both the printer and the OctoPrint server.  Sadly, it doesn't give an error of any type, and the logs don't mention anything.

donovan6000, 2016-07-29:
Someone just reported something similar on M3D's forum. Seems like something changes in how OctoPrint handles communication timeouts. I'm looking into it now.

donovan6000, 2016-07-29:
Just released [M33 Fio V1.4](https://github.com/donovan6000/M33-Fio/releases/tag/V1.4) that fixes this.

Alaric323, 2016-07-29:
Nice changelog. :P Testing it out now.

Alaric323, 2016-07-30:
Froze in the middle of a print.  Here's the log for when it happened.
[octoprint (2).txt](https://github.com/donovan6000/M33-Fio/files/391622/octoprint.2.txt)

donovan6000, 2016-07-30:
The log doesn't show any reason why it froze, so it was probably a firmware issue. If your using M3D's V2016040401 firmware then downgrade to M3D's V2016010812 firmware since the former has been known to cause this issue occasionally. The newest version of iMe firmware hasn't been tested much, so it could also potentially have caused it to freeze.

Alaric323, 2016-07-30:
Was on iMe 1.06, downgraded to the January 2016 firmware and am printing again.  I'll update what happens.

donovan6000, 2016-09-21:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2016-09-23 by donovan6000