donovan6000/M33-Fio #179: How to determine if layer shift is hardware, firmware, or M33-fio/Octaprint

scallviz, 2016-08-09:
Hi!

After having poor luck since launch with my M3D (periods of mostly functional and then long stretches of nothing quite working) I found your project.

I have successfully installed iMe on my M3D and Octaprint+M33-Fio on my macbook pro.

In my testing, almost every model I print has a layer shift.  Usually Y, but every now and then X.

Sometimes it shifts, prints a few layers, and then shifts back, sometimes in just shifts and stays shifted until the end of the print (or I cancel it)

I was just wondering if there's an easy way to determine if it's hardware, firmware, or software that might be causing the shifts?  I saw in the issues that there was a rounding error (in iMe?) that was causing some shifts but was corrected.

Thank you for all of your hard work on these projects and any advice you can share would be great.

Thanks!

donovan6000, 2016-08-10:
The shifts are most likely the hardware's fault. The motors that control each of the axis have a max speed that they can move, and the motors will skip steps and causes shifts if that max speed is exceeded. The firmware enforces a max speed limit for each of the motors to fix this problem, however every printer is a little bit different due to wear so the speed limit for your printer probably will be unique. Try decreasing the max X and Y speed limits to something like 1000mm/min to see if that helps. You can change this setting in the M33 Fio tab in OctoPrint's settings.
[image]
Also the way that the filament is fed to the extruder can cause shifts it the filament tugs on the print head too much. So make sure that there's not too much tension on the filament as it's being fed to the extruder.

qwerty8224, 2016-08-12:
I have a problemmu slowed to 720mm.min shift there.

qwerty8224, 2016-08-12:
720 mm.min have offset

donovan6000, 2016-08-13:
@qwerty8224 It does seem kinda weird that your still getting layer shifts at that slow speed. Try out iMe's latest devel version, [V00.00.01.09](https://github.com/donovan6000/iMe/tree/devel), and let me know if your still experiencing the shifting issue.

Does printing the same file multiple times with the same settings result in the displacement happening at the same location? If it does, then can you upload the G-code file that your printing?

qwerty8224, 2016-08-13:
Printing stops immediately. IME1.09

donovan6000, 2016-08-13:
What printing software are you using? I've been testing it on Cura 15.04 and OctoPrint, and it starts printing without any problems. Was the printer responsive at all after the print stopped or did the printer just crash?

qwerty8224, 2016-08-13:
M33Fio Movement is to big ?

scallviz, 2016-08-14:
I'm seeing the same issue with 1.09 (iMe 1900000109.hex) it started to print and then froze.  I tried it after getting shifting (ever worse?) when I turned the speed from 1500 1000 on the axes.

Octoprint 1.2.15 with M33-Fio 1.5 and Cura Engine (<=15.04) 

Thanks!

scallviz, 2016-08-14:
I also noticed the Micro 3D Profile in octoprint has 4800 mm/minute, does that affect anything or does the 1000 mm/minute firmware setting take precedence?

donovan6000, 2016-08-16:
The speed settings in the printer profile aren't used by anything in M33 Fio, and I don't think OctoPrint uses them for anything either. It's just included for completeness. The speed that is used when printing is initially decided by the settings provided to the slicer that slices the the 3D model, but the firmware prevents that speed from being outside of a certain range. So the firmware speed limits take precedence.

Try out the newest devel version of iMe, [V00.00.01.10](https://github.com/donovan6000/iMe). It should fix the freezing issue in the last devel version. And let me know if printing the same file multiple times with the same settings result in the displacement happening at the same location on the same layer.

qwerty8224, 2016-08-16:
speed of 720 mm / min bias NO
tested MORE

qwerty8224, 2016-08-17:
Print speed infill16 
insade 14
outside12
travel 20 
OK
I tested on.....

qwerty8224, 2016-08-17:
Print a detail of all SUPER

scallviz, 2016-08-30:
So I thought my layer shifting might be due to the M3D needing lubrication.

Following instructions on the M3D site I applied Teflon silicon lubricant to the rods (and not the belts) .

Sure enough, the layer shifting went away.

However I'm getting a new phenomenon which I'm thinking might be a sign that I need to redo my backlash.

Here is a picture of a simple cylinder printed after the lubrication:
https://goo.gl/photos/xJ9AybhJ9NgymNNg9

The work I did on the printer coincided with the release of FIO 1.6, so I installed that and the .12 iMe firmware so I'm wondering if something in the .12 firmware changed as well?

Also if there's a better venue (message boards or something) to communicate, let me know I don't mind github but don't want to abuse it either :)

Thanks

donovan6000, 2016-08-30:
It looks like you need to calibrate your backlash more accuratley. M33 Fio provides an easy way to print [Muele's backlash objects](https://www.thingiverse.com/thing:1435828) and M3D has another method for calibrating it(https://printm3d.com/solutions/article.php?id=234).

[Here's the changelog for iMe](https://raw.githubusercontent.com/donovan6000/iMe/master/Changelog). It shows everything that's changes in each new version, and there's also a [devel version of iMe](https://github.com/donovan6000/iMe/tree/devel) that has all the stuff we're testing out for the next release.

I tend to use [M3D's forum](https://printm3d.com/community/viewtopic.php?f=5&t=11490) and the [M3D Thingiverse group](https://www.thingiverse.com/groups/m3d-the-micro-3d-printer/topic:5096), but I don't really mind using Github for discussing the problems with M33 Fio and iMe.

donovan6000, 2016-09-21:
I'm closing this issue now since the layer shifting issue has been fixed. Feel free to open another issue or use the forums that I mentioned if you have any other problems.

closed 2016-09-21 by donovan6000