donovan6000/M33-Fio #185: external slicing

analogmonster, 2016-09-22:
I am interested in comparing a few different slicers using octoprint and m33fio.  I would like to know what m33fio does if you send octoprint gcode instead of slicing using cura engine.  What about backlash and bed compensation?  I am using ime firmware.

I need to understand quite more about quite a few of the options.  what is pre process on the fly?  Is validating just checking it is within bounds and speed limits?  Why would I remove temperature and fan commands? 

Also I am using an external bed.  Does adjusting the external bed height parameter account for the reduced print size or is the bound just shifted up?

donovan6000, 2016-09-23:
All of M33 Fio's settings and buttons have tooltips that appear when you hover your mouse over them for a few seconds. That should help you better understand what everything does.
[image]

Pre-processing on the fly allows each command in a G-code file being printed to be passed through the pre-processor stages immediately before that command is sent to the printer. That allows the G-code file to remain unchanged, and all settings' current values will be used when printing that G-code file. Without on the fly pre-processing, all G-code files sent to OctoPrint or sliced from within OctoPrint are sent through the pre-processor stages as soon as they are finished being sent to OctoPrint or finished being sliced. That permanently changes the G-code file to use the current values of all the settings. Any settings changed after that G-code file has been pre-processed wont affect that file in any way

M3D's official software doesn't use on the fly pre-processing, and that's why it makes you slice a file every time you want to print it. With M33 Fio's on the fly pre-processing, you only have to slice a file once. On the fly pre-processing should almost always be enabled. The only times I disable it are for seeing how the pre-processor stages affect a file when I'm debugging M33 Fio.

M33 Fio's pre-processor stages will do different things depending on the type of firmware your using, such as not applying the bed and backlash compensation when using iMe firmware. Thanks to on the fly pre-processing, M33 Fio can detect your printers firmware and not use the inapplicable parts of the pre-processor stages when your printing a file.

M33 Fio does the same thing to G-code being sent to OctoPrint as it does to G-code being generated by slicers from within OctoPrint. If you have on the fly pre-processing enabled, then M33 Fio doesn't do anything to the G-code file since all the changes are applied as the G-code is being printed as I explained before.

The Validation pre-processor stage removes commands that the Micro 3D printer doesn't understand, such as T0, M84, G21, etc.

The Preparation and Thermal bonding pre-processor stages add in commands that set the fan speed and extruder temperature depending on the temperature and filament type settings, so choosing to remove temperature and fan commands make it so that any existing fan and temperature commands don't interfere with those added by the Preparation and Thermal bonding pre-processor stages. If your sending a file to OctoPrint that already contains the fan speed and extruder temperature that you want to use, then you can disable the remove fan and temperature commands settings to keep the fan speed and extruder temperature that you want to use. If your doing that then you should probably also disable the Preparation and Thermal bonding pre-processor stages so that the commands added in by them don't override your desired temperature and fan commands.

The external bed height accounts for the reduced print area. You can visually see the affects of changing the external bed height in M33 Fio's model editor. For example, here's what the print area looks like with an external bed height of 30mm.
[image]

analogmonster, 2016-10-03:
awesome thanks for all of that

closed 2016-10-03 by analogmonster

donovan6000, 2016-10-03:
Glad I could help. I know that M33 Fio doesn't really have any detailed documentation, so feel free to open another issue if you ever have anymore questions.