donovan6000/M33-Fio #188: Print head losing position when print ends

ctday, 2016-10-01:
My Raspberry Pi 3 running OctoPrinter and M33 Fio with iMe firmware, lastest versions all round, loses track of the printhead's position after a print ends, either naturally or on a Cancel. If I immediately start another print, it starts way out of position, usually far to the left and fairly randomly in Y. It seems to take at least a bed calibration or two for the system to figure out where it actually is. Any ideas?

donovan6000, 2016-10-02:
Is this being caused by the X/Y jerk homing being too sensitive? iMe's homing algorithm uses the accelerometer to detect when the extruder hits the back and right walls, and this detection can fail depending on the jerk sensitivity. Try setting the jerk sensitivities to something really low, like 1, in the M33 Fio tab of OctoPrint's settings to see if that's what's causing the problem.

donovan6000, 2016-11-21:
It's been awhile since anyone's responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in fixing/testing this problem.

closed 2016-11-24 by donovan6000