donovan6000/M33-Fio #189: Printer didn§

ra100, 2016-10-02:


ra100, 2016-10-02:
Update: this happens with test border too.

donovan6000, 2016-10-13:
Sorry it took so long to get back to you. Try out the [newest devel version](https://github.com/donovan6000/M33-Fio/tree/devel) and let me know if you still experience this issue.

ra100, 2016-10-13:
It doesn't look like it helped. I tried to print test border and when the print finished. Heater kept heating and I even button to turn heater off didn't work, I need to use "Print" button and then "Cancel".
```
Send: N221 G0 E144.3689*86
Recv: ok 221
Send: N222 G0 X-0.5 Y8 Z0.4 E145.8858*50
Recv: ok 222
Send: N223 G0 X-2 Y6.5 Z2.4 E151.1951*57
Recv: ok 223
Send: N224 G0 X-2.000000 Y6.500000 Z1.817299 E151.1951 F900**110
Recv: ok 224
Send: N225 G0 E148.1951*86
Recv: ok 225
Changing monitoring state from 'Printing' to 'Operational'            <--- Print ended
Send: N1 G4*44
Send: M114
Recv: T:252.6694
Recv: T:254.3300
Recv: T:249.2017
Recv: T:243.3175
Recv: T:241.1385
Recv: T:240.3774
Recv: T:235.1135
Send: N0 M110*35
Recv: ok  X:-2.0005 Y:6.4968 E:148.1951 Z:1.8171
Recv: T:238.6144
Recv: T:238.6144
Recv: T:238.6144
Recv: T:239.3256
Recv: T:239.3256
Recv: T:239.3256
Recv: T:240.7529
Recv: T:239.3256
Recv: T:239.3256
Recv: T:238.6144
Recv: T:239.3621
Recv: T:240.0384
Recv: T:238.6500
Recv: T:238.6144
Recv: T:240.7148
Recv: T:237.9047
Recv: T:239.3621
Recv: T:240.7148
Recv: T:240.7148
Recv: T:238.6144
Recv: T:239.2893
Recv: T:240.0012
Recv: T:240.0758
Recv: T:238.6144
Recv: T:238.6500
Recv: T:238.6500
Recv: T:239.3621
Recv: T:239.3621
Recv: T:240.7148
Recv: T:239.3621
Recv: T:239.3256
Recv: T:238.6144
Recv: T:239.3256
Recv: T:238.6500
Recv: T:238.6144
Recv: T:239.3256
Recv: T:240.0012
Recv: T:239.3256
Recv: T:239.3256                                          <---- heater kept heating
Recv: ok 0                                                <---- pressed the Print button
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok 0
Send: N1 G90*17
Recv: ok 1
Send: N2 M106 S1**78
Recv: ok 2
Send: N3 M17**60
Recv: ok 3
Send: N4 G90**62
Recv: ok 4
Send: N5 M104 S240**76
Recv: ok 5
Send: N6 G0 Z5.000000 F48**14
Recv: ok 6
Send: N7 G28**62
Recv: T:249.1530
Recv: T:241.8632
Recv: T:238.9744
Recv: T:241.8632
Recv: ok 7
Send: N8 M18**56
Send: N9 M109 S240**77
Changing monitoring state from 'Printing' to 'Operational'
Send: G4 S15*
Recv: T:240.3394
Send: M106 S255*
Send: M17*
Send: G92 E0*
Send: G0 Z0.400000 F48*
Send: G90*
Warn: Not sending "M140 S0", printer profile has no heated bed
Send: M84
Send: M104 T0 S0
Send: M106 S0
Recv: wait
Recv: T:225.8915
Send: N0 M110*35
Recv: ok 0
Send: M114
Send: N1 G4*44
Send: N0 M110*35
Recv: ok  X:53.9965 Y:49.9962 E:0.0000 Z:5.0001
Recv: ok 1
Recv: ok 0
Send: N1 M104 S0*100
Send: N2 G90*18
Recv: ok 1
Recv: T:0
Send: N3 G0 Y105.000000 Z15.000100 F1800*119
Recv: ok 2
Send: N4 M107 T1*100
Recv: ok 3
Send: N5 M18*31
Recv: ok 4
Send: N6 M107*35
Recv: ok 5
Recv: ok 6
Send: N7 G4*42
Recv: ok 7
Send: N8 M65539
Recv: ok 8
Send: N9 M65541
Recv: wait
```

ra100, 2016-10-15:
I am not sure what changed, but it is fixed. Thanks.

donovan6000, 2016-10-15:
I didn't change anything related to it, so I'm not sure why it's fixed. I'm still not sure what's causing it since it never seems to fail performing the standby actions for me. Let me know if you ever expirence this problem again.

ra100, 2016-10-16:
Probably I just didn't wait long enough.

ra100, 2016-10-19:
I am maybe closer to what went wrong. I just tried printing with OctoPrint 1.3.0rc1 and this issue occured, when I switched back to 1.2.x, it is OK.

donovan6000, 2016-10-20:
Thanks. I can confirm that the standby commands aren't working on OctoPrint 1.3.0rc1. I'll look into fixing this in the next few days.

ra100, 2016-10-22:
When you'll need to check if it works, let me know.

donovan6000, 2016-10-25:
I think I've got it working now in the devel branch if you want to test it :)

ra100, 2016-10-25:
Awesome, it works.

closed 2016-10-25 by ra100