donovan6000/M33-Fio #190: M33 Calabration problems

bobbypop999, 2016-10-09:
When I Calabration my printer and print an object it's fine but I go to print again and the Z axis is too low or too high and very high like half way up. I am using the latest m3d file are and latest software. 

Please could you help.

donovan6000, 2016-10-10:
It sounds like there could potentially be a problem with your printer. Does this happen when using a different firmware, like M3D V2016010812 or iMe V00.00.01.22?

Also, you can enable the 'Calibrate before print' setting in the M33 FIo tab of OctoPrint's settings to have your printer calibrate its bed center Z0 before every print.

bobbypop999, 2016-11-04:
I hope not I will try soon. Sorry for not getting back quick enough haven't had time to test.

bobbypop999, 2016-11-19:
It seem fine now.

closed 2016-11-19 by bobbypop999