donovan6000/M33-Fio #191: Install creates m3dfio files alongside m33fio

ra100, 2016-10-15:
This is isssue since m3dfio was changed to m33fio, I needed to disable m3dfio plugin in octoprint config, to be able to run this plugin.
Now I noticed, that when I run `python setup.py install`, on last lines I see:
```
copying OctoPrint_M33Fio.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying OctoPrint_M33Fio.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying OctoPrint_M33Fio.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying OctoPrint_M33Fio.egg-info/entry_points.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying OctoPrint_M33Fio.egg-info/not-zip-safe -> build/bdist.linux-armv7l/egg/EGG-INFO
copying OctoPrint_M33Fio.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
```

Octoprint startup, with disabled m3dfio
```
|  M33 Fio (1.8.2016101219) = /usr/local/lib/python2.7/dist-packages/OctoPrint_M33Fio-1.8.2016101219-py2.7.egg/octoprint_m33fio
| !M3D Fio (0.17) = /usr/local/lib/python2.7/dist-packages/OctoPrint_M33Fio-1.8.2016101219-py2.7.egg/octoprint_m3dfio
```

donovan6000, 2016-10-18:
[Setup.py now uninstalls M3D Fio if it can find pip](https://github.com/donovan6000/M33-Fio/commit/983a44be89797f44644a581c6b652d4918c96c7d). That should resolve all the issues that would happen by having both M3D Fio and M33 Fio installed at the same time.

I tried looking into having M33 Fio disable or uninstall M3D Fio, but nothing seemed to work or it required restarting OctoPrint to take affect. So I'm satisfied with having setup.py take care of it all.

ra100, 2016-10-22:
This definitely solved my problem. Thanks.

closed 2016-10-22 by ra100