donovan6000/M33-Fio #194: external extruder

analogmonster, 2016-10-25:
I am really fed up with my M3D Micro.  I've had a new extruder motor fitted for a few weeks.  Now I have the clear cover I can actually see the filament slip on longer prints as it gets warm.  The design is fundamentally flawed.  I was thinking about trying to find a replacement stepper with better specs and keeping it direct drive, but I am now pretty certain I'm going to install a bowden.  That way I can get the hot end away from the extruder and get NEMA steppers instead of these crappy plastic geared steppers that were only designed to open curtains and blinds - hardly a use case where accuracy, repeatability or torque are required.

So I want to use an E3D Titan extruder, possibly with a mechaduino or ustepper closed loop driver. [http://ustepper.com](url) or [http://tropical-labs.com/index.php/mechaduino](url).  Haven't settled on that yet, but I have some here to test out so will see if it adds anything. I know there are DRV8834s in there but with no heat sink and a low confidence in their design skills it would make me less concerned about the control board struggling with the increased load too.  

The Titan WIKI [http://wiki.e3d-online.com/wiki/Titan_Assembly](url) mentions how to calculate the steps per unit.  am I right in thinking I can just plug that data into the M33-Fio steps/mm box if I am running the iMe firmware?  Also what are the defaults for the X/Y/Z steps/mm?  Thanks

donovan6000, 2016-10-26:
I completely agree that the cheap extruder motor and its close proximity to the heater are probably the worst things about the printer. I have seen some people change the motor [here](http://www.instructables.com/id/M3D-Printer-Setup-External-Extruder/) and [here](https://plus.google.com/+AndreaMunegato/posts/4UdLi1K2DQc), so it's certainly possible to do what your trying to accomplish.

Your correct about just having to plug the new steps/mm into M33 Fio's settings to apply them. iMe will read the steps/mm settings from its EEPROM every time a motor is used, so changing the steps/mm settings will immediately take affect.

There's also an E motor current settings, but it could potentially damage the circuit board so I wouldn't recommend changing it. According to M3D's official software, the two extruder currents that they use are 500mA and 660mA, however the extruder current setting in EEPROM can be change to allow the current to be anywhere, within reason, from 0mA to 65535mA. All printers with serial numbers that start with the following are forced by the software to have a 500mA extruder current. Those with different starting serial numbers aren't limited by the software so they can have any extuder current, but since M3D has only ever used those two current values it'll either have a current of 500mA or 660mA.
```
BK15033001100
BK15040201050
BK15040301050
BK15040602050
BK15040801050
BK15040802100
GR15032702100
GR15033101100
GR15040601100
GR15040701100
OR15032701100
SL15032601050
```

All the default steps/mm are:
```
#define EEPROM_X_MOTOR_STEPS_PER_MM_DEFAULT 19.3067875
#define EEPROM_Y_MOTOR_STEPS_PER_MM_DEFAULT 18.00885
#define EEPROM_Z_MOTOR_STEPS_PER_MM_DEFAULT 646.3295
#define EEPROM_E_MOTOR_STEPS_PER_MM_DEFAULT 128.451375
```

You could add some heatsinks to the motor controllers if your worried about pushing them too hard. The 4 DRV8834s are the chips on the right of the board, and there's a little bit of open space on top of each of them to put a heat sink.
[image]

analogmonster, 2016-10-26:
Thanks for the defaults, I've fiddled with the settings so much that I wanted to be sure I hadn't changed anything in that area.

Thanks for the links too. I'd seen the instructables one but not the other. That has only reinforced that a Bowden is needed since he's using the stock stepper and getting good results. Still think I want to swap for a stepper with a NEMA simply to get away from plastic cogs and hoping it should retract better which will be key for getting the Bowden right.

The Titan comes with a NEMA 17 40N stepper, current isn't listed on the site but I'd expect a little more than 1A. drv8834s can do that, but above 1.5A and they definitely will need heat sinking and that depends on the board layout being thermally optimised to begin with. 

I'm going to invest in an infrared thermometer and try driving it from stock first, consider a heat sink or external driving only if required. Thanks for the input I think you've given me everything I need at the M33-Fio end to get this working

closed 2016-10-26 by analogmonster